OrcaSlicer/OrcaSlicer #2365: Creating profiles from scratch

ParadiseFallen, 2023-10-08:
I want to have ability add profile from scratch. Right now im bounded to `system` profiles. And i need to build on top of them mine.
It would be nice to add ability to make custom root profile and inherit them later.

Same with filaments. Need ability to add pack of filaments when we can craete our own and use em across printers / projects.

Processes must be bounded to printer but also created and modified.
Right now we cant modify build it presets.

closed 2024-01-15 by github-actions[bot]

Apotrox, 2024-02-28:
+1, i just want to create filament profiles. Not tie them to anything. maybe being able to create copies for different colors/variants. Don't make me tie them to a printer or system preset. very silly.

ParadiseFallen, 2024-08-31:
reopen