OrcaSlicer/OrcaSlicer #2367: Smart Auto Upload and Start Print

alfondehesa, 2023-10-08:
**Is your feature request related to a problem? Please describe.**
Currently, OrcaSlicer uploads gcode directly to octoprint. This workflow could be much more streamlined.

**Which printers will be beneficial to this feature**
All printers with octoprint as the network interface.

**Describe the solution you'd like**
OrcaSlicer should upload GCODE to Octoprint, wait until the printer turns on and is connected (or if not, turn it on), wait for GCODE analysis to complete, and then start the print.

**Describe alternatives you've considered**
[This plugin for Cura](https://github.com/fieldOfView/Cura-OctoPrintPlugin) performs this specific workflow.

**Additional context**
None.

Ocraftyone, 2023-11-04:
If you are using klipper, Moonraker has some powerful power settings. You can specify that any time a file is added to the queue, it turns on the printer, waits to connect to the printer, then starts the job. https://moonraker.readthedocs.io/en/latest/configuration/#power

evilDave, 2024-02-07:
It would be nice to see something like this feature make it possible to use "upload and print" with post processing plugins active on octoprint.

E.g. I have ArcWelder plugin processing incoming files, and "upload and print" has no way to wait for the processing to be complete: the original file gets printed, OR the file cannot be overwritten because it is already printing.

Something like detecting the processing complete and triggering the print of the same or updated file, or at the very least specify a delay between upload and print.

closed 2024-05-15 by github-actions[bot]