OrcaSlicer/OrcaSlicer #2369: G28 homing issue at start print causing print to be started at the end of the bed and over running y axis

BSKFeralLycan, 2023-10-09:
### OrcaSlicer Version

1.7.0

### OS version

windows 11

### Additional system information

Raspberry pi 4 8gb ram running octo print os

### Printer

Sovol SV06+

### How to reproduce

1. set up SV06+ profile ran my start gcode from prusa unstead of default here
2.  - G90 ; use absolute coordinates
M83 ; extruder relative mode
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
{if filament_settings_id[initial_tool]=~/.*Prusament PA11.*/}
M420 S ; enable mesh leveling
G1 Z0.35 F720
G1 Y-3 F1000 ; go outside print area
G92 E0
G1 X60 E9 F1000 ; intro line
G1 X100 E9 F1000 ; intro line
{else}
G1 Z0.2.5 F720
G1 Y-3 F1000 ; go outside print area
G92 E0
G1 X60 E9 F1000 ; intro line
G1 X100 E12.5 F1000 ; intro line
{endif}
G92 E0
M221 S{if layer_height<0.075}100{else}99{endif}
M300 P200
3. made a calibration temp tower
4. uploaded to octopi
5. started print

### Actual results

6. machine at homing ran bed forward over about 150mm and stopped
7. extruder moved to middle of bed purged filament 
8. resumed to move to the rear of the print bed and start print 
9. I thought this was an issue with starting gcode erased  and used default gcode with same result but purge line in the middle of the med all the way to the right for almost 150mm way longer then normal
10. went back to orca slicer to make sure bed size and origin were correct at 310x310x340 and origin at (0,0) returned to print
11. stopped print. hit g28 in terminal on octo print within orca slicer thinking it would bring my bed back to the middle
12. extruder homed but bed just tried going forward again and stopped; extruder then came down on temp tower twisting x axis and breaking my ABL (my fault should have removed the print but correct homing would not have interfered with it at the rear at all) emergency stopped reset removed print tightened probe
13. hit g28 again bed didn't move extruder homed to the middle and straight into my bed again
14. took off WLAN and octopi connection reset pi sent g28 code from pc interface and not in orca printer homed as usual. 
15. after attempting to fix my machine i sent a file through prusa directly to octopi hit print and printed as normal as it could with a now warped bed and x axis rails

After looking for help on groups finding others having the same issue one even with the voron, i hit the forums here expecting answers but none could be found so i started typing

### Expected results

when we hit print it should properly home the axis's and print where the model is positioned on the plate in orcaslicer.

### Project file & Debug log uploads

[logs.zip](https://github.com/SoftFever/OrcaSlicer/files/12842658/logs.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

blueshockz47, 2023-10-18:
did you solve this issue?I am having problem also on my Neptune 3 Max with klipper. Prusa Slicer works fine.

tusculumgolfer, 2024-01-11:
Running into same issue. Klipperized SV06 Plus

closed 2024-04-19 by github-actions[bot]

olaab, 2024-04-23:
And what was the solution?
Have the same problem with OrcaSlicer V2.0.0
Win 10
Hello out there!