OrcaSlicer/OrcaSlicer #2375: USB device eject doesn't work in macOS Ventura

martin-badin, 2023-10-09:
### OrcaSlicer Version

1.7.0

### OS version

macOS Ventura 13.6

### Printer

Elegoo Neptune 3 Plus

### How to reproduce

1. Import any STL file
2. Click Slice button
3. Click Export G-code file
4. Try to eject USB device

### Actual results

USB device eject doesn't work

### Expected results

USB device eject should work

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2023-10-12:
Doesn't work in Manjaro either.

Furthermore, if you try to eject the drive using the dedicated button after saving a g-code that was already on the disk **(Content is irrelevant, the name matters!)**, its content will be erased! 😲 🤷🏻‍♂️

enotskellington, 2023-12-19:
I haven't found that it is erased (on Debian 12/Gnome) but I do find that eject seems to hang and nothing happens. For now I have switched to 'open in folder' > eject from there.

discip, 2023-12-23:
@SoftFever @Noisyfox

Any solution for this yet?

orgoj, 2023-12-24:
On Lubuntu eject doesn't work on. Prusa slicer eject is fine.

SoftFever, 2023-12-24:
will take a look,
Haven't been using SD card for years, lol

enotskellington, 2023-12-24:
> [Tracking image][https://github.com/notifications/beacon/A5P3TGKEMMYNWK6EDUH3QRLYK74ENA5CNFSM6AAAAAA5Y6DZLSWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTPL234G.gif]

I guess I should add this is an SD card using a USB adapter. But as others mentioned PS works.

discip, 2024-01-18:
Any progress so far? 😊

discip, 2024-01-22:
@SoftFever
Any progress so far? 😊

discip, 2024-02-19:
@SoftFever
@Noisyfox
@igiannakas
Has anyone looked into this yet?
Would be great if this could finally be solved.

myxxmikeyxx, 2024-05-06:
I am also having this problem on MacOS Sonoma 14.2.1 (23C71) (Intel).
It just keeps saying ejecting but never does.

myxxmikeyxx, 2024-05-07:
@discip I had not tried it. I downloaded the nightly build and it did work for ejecting the usb! Thank you!

discip, 2024-05-07:
You are welcome.
However, you should thank the guys who fixed this. 😊

discip, 2024-05-07:
@martin-badin
As this seems to be fixed, please test and if it works for you as well, please don't forget to close. 😊

martin-badin, 2024-05-07:
@discip Thank you. Currently, I am using wireless connection. So I believe it works. I am closing this issue.

closed 2024-05-07 by martin-badin