OrcaSlicer/OrcaSlicer #2384: Crash when trying to use pressure avance calibration

kerbeus-a, 2023-10-10:
### OrcaSlicer Version

1.7

### OS version

Windows 10 Enterprise

### Printer

Voron Trident (happens on other printers too)

### How to reproduce

1. go to Calibration
2. select Pressure Advance
3. Select PA Pattern
4. click OK

### Actual results

an alert OrcaSlicer pop up oppens with the following error message
Unhandled, unknown exception, terminating the application

### Expected results

produce a PA pattern for slicing/sending to printer

### Project file & Debug log uploads

[crash_Mon_Oct_02_15_41_02_0.zip](https://github.com/SoftFever/OrcaSlicer/files/12860313/crash_Mon_Oct_02_15_41_02_0.zip)
[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

Noisyfox, 2023-10-13:
Can you try the latest dev version and see if it still crashes? If it does, could you post the new crash report?

kerbeus-a, 2023-10-17:
> Can you try the latest dev version and see if it still crashes? If it does, could you post the new app crash log?

I can definitely do that. where do I find the latest dev release?

Noisyfox, 2023-10-17:
> > Can you try the latest dev version and see if it still crashes? If it does, could you post the new app crash log?
> 
> I can definitely do that. where do I find the latest dev release?

https://github.com/SoftFever/OrcaSlicer/actions/runs/6523422600
Try this one.

kerbeus-a, 2023-10-17:
> > > Can you try the latest dev version and see if it still crashes? If it does, could you post the new app crash log?
> > 
> > 
> > I can definitely do that. where do I find the latest dev release?
> 
> https://github.com/SoftFever/OrcaSlicer/actions/runs/6523422600 Try this one.

tried the provided 1.8 dev one - same result/error.  
I did use the portable version both times if it changes anything

kerbeus-a, 2023-10-17:
[debug_Tue_Oct_17_09_41_52_20320.log.zip](https://github.com/SoftFever/OrcaSlicer/files/12929656/debug_Tue_Oct_17_09_41_52_20320.log.zip)
here is the log

Noisyfox, 2023-10-17:
Did it generate the crash_xxxxx.log? Could you upload that as well? Thanks.

kerbeus-a, 2023-10-17:
> Did it generate the crash_xxxxx.log? Could you upload that as well? Thanks.

there were no crush log. I tried a few times.  Also - when the Error pop up appears and I click OKon it, the app closes. But it is possible to ignore the pop up and click on the main window. The pop up goes back behind the main window. I can click Ok on the PA test again and the Save pop up appears. I ignore the save and then it starts calculating and is very slow. the rest of the system is responsive but the Orca slicer is slow. 
[image]

if I repeat clicking OK on the PA test the pop up appears and disappears fast but I am able to reat the Heading : Creating a new project: unsaved changes.  The PA dialog still shows
[image]

I can rotate the build plate etc but it is choppy and very  very slow, Like so : I drag the buildplate around and it jumps into new position a few seconds later. there is no affect on my CPU usage etc. the rest of the system is fast as usual

kerbeus-a, 2023-10-17:
also - I just found - if I dont click OK on the error pop up but just close it it generates the crash log.
[crash_Tue_Oct_17_14_01_23_0.zip](https://github.com/SoftFever/OrcaSlicer/files/12937792/crash_Tue_Oct_17_14_01_23_0.zip)

Noisyfox, 2023-10-18:
Did it only crash with certain printer profile?

kerbeus-a, 2023-10-18:
interesting. it produced an error on all the profiles I had but they were all imported from superslicer. I tried a system voron profile and there is no issue

Noisyfox, 2023-10-18:
Could you also share those profile files as well? You can just zip all the files from "Orca config folder/user"

kerbeus-a, 2023-10-18:
> Could you also share those profile files as well? You can just zip all the files under "Orca config folder/user" and upload it here, or email me.

profiles emailed.

Noisyfox, 2023-10-19:
Could you also send me the 'Orca config folder/system' as well? Thanks.

kerbeus-a, 2023-10-20:
> Could you also send me the 'Orca config folder/system' as well? Thanks.

for sure. 
sent

Noisyfox, 2023-10-22:
> > Could you also send me the 'Orca config folder/system' as well? Thanks.
> 
> for sure. sent

Cool I'm able to reproduce the problem with your presets now. Will try to figure out what's going on.

Noisyfox, 2023-10-22:
Ok the problem is that those profiles have the default line width set to 0. Change it to a positive value should fix the problem.

kerbeus-a, 2023-10-23:
thank you! got it

silence48, 2023-12-31:
This is marked resolved but the logs show another exception Code :c0000005 ACCESS_VIOLATION

thus I'm adding it to the list of related issues:

[2921](https://github.com/SoftFever/OrcaSlicer/issues/2921) 
[3375](https://github.com/SoftFever/OrcaSlicer/issues/3375) 
[3157](https://github.com/SoftFever/OrcaSlicer/issues/3157)
[3340](https://github.com/SoftFever/OrcaSlicer/issues/3340)
[3292](https://github.com/SoftFever/OrcaSlicer/issues/3292)
[2896](https://github.com/SoftFever/OrcaSlicer/issues/2896)
[2766](https://github.com/SoftFever/OrcaSlicer/issues/2766)
[2888](https://github.com/SoftFever/OrcaSlicer/issues/2888)
[3174](https://github.com/SoftFever/OrcaSlicer/issues/3174)
[2803](https://github.com/SoftFever/OrcaSlicer/issues/2803)
[2484](https://github.com/SoftFever/OrcaSlicer/issues/2484)

knownchild, 2024-02-24:
works in bambu slicer...

WinstonLeWolf, 2024-03-17:
This issue still persists and changing the default line width fixes it.  Is it possible to add a error dialog or toast notification that informs you that the default line width needs to be a positive value?

closed 2024-07-12 by github-actions[bot]