OrcaSlicer/OrcaSlicer #2388: Mouse Ears fill the inside of Text

Fragmon, 2023-10-11:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Additional system information

non hardware related

### Printer

Voron 2.4 350mm

### How to reproduce

Slice an object with Text.
Activate Mouse ear

### Actual results

The text will be filled with brim. If you have only one layer text, you cant remove that.

### Expected results

No mouse ears in the Text

### Project file & Debug log uploads

[image]
[image]
[debug_Tue_Oct_10_18_32_23_11072.log](https://github.com/SoftFever/OrcaSlicer/files/12865673/debug_Tue_Oct_10_18_32_23_11072.log)
[gehause_oberteil-v39.zip](https://github.com/SoftFever/OrcaSlicer/files/12865676/gehause_oberteil-v39.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2023-10-11:
Currently brim ears are applied to both inner and outer region. I was thinking if we should disable brims for any single-layer (or maybe more than 1 layer?) regions to avoid brim fusing with bridges.

Fragmon, 2023-10-11:
> Currently brim ears are applied to both inner and outer region. I was thinking if we should disable brims for any single-layer (or maybe more than 1 layer?) regions to avoid brim fusing with bridges.

Would it be possible to introduce a tool, similar to the support feature, that allows users to designate areas where a brim shouldn't be generated?

Noisyfox, 2023-10-11:
I think it's possible to add some sort of modifier that prevents creating brims.

Fragmon, 2023-10-11:
It's puzzling that when 'auto brim type' is selected, Orca doesn't generate it internally.
[image]

Noisyfox, 2023-10-11:
Yeah I know it's a bit puzzling.

Noisyfox, 2023-10-11:
Oh and by the way, auto brim will NEVER generate inner brims.

Fragmon, 2023-10-11:
> Oh and by the way, auto brim will NEVER generate inner brims.

Can i deactivate inner brims with mouse ears?

Noisyfox, 2023-10-11:
Not at the moment. Will consider allowing so.

closed 2024-01-17 by github-actions[bot]

doudar, 2024-04-04:
Id like this re-opened because I have the same issue with nylon prints, where I have recessed text and need mouse ears to keep the nylon from warping off the plate. Brim works, but it's very difficult to remove from nylon prints.

crookz, 2024-04-06:
same, it makes ears useless for most of my current prints (with text)

icariantk, 2024-04-25:
+1 i just need the ears only on a single part of the object, a blocking cube could be enought to do that at least.