OrcaSlicer/OrcaSlicer #2390: Mouse ear detection radius strange behaviour (lower right corner)

CooperGerman, 2023-10-11:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Additional system information

Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz   3.40 GHz
RAM 32,0 Go
Type du système	Système d’exploitation 64 bits, processeur x64

### Printer

Voron trident 250

### How to reproduce

1. slice the attached test file with attached project file without alterations
2. verify that the lower right mouse ear is missing
3. change `Brim ear detection radius` from `1` to `2`
4. slice again and verify all 4 mouse ears are present

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[mouse_ear_test.zip](https://github.com/SoftFever/OrcaSlicer/files/12867642/mouse_ear_test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2023-10-11:
This is a known issue and there is no good solution. This is caused by some sort of rounding error so that even though the 4 corners look the same to eyes but they are slightly different.

CooperGerman, 2023-10-11:
Oh ok, i wasn't aware of that. Since the radius was 1.6 mm I imagine that it is rounded to nearest integer ? Felt strange that with a threshold at 1 and a 1,6mm radius this occurs ... thx for the quick answer though.

Noisyfox, 2023-10-12:
I'm currently experimenting on improving the mouse ear detection on rounded corners by utilizing the arc-fitting algorithm and hopefully can solve your problem.

closed 2024-01-19 by github-actions[bot]

StephanCassel, 2025-06-04:
Using Mac and OrcaSlicer 2.3.0:
Same problem. Preview only 3 ears but that also go for the print

[image]

.