OrcaSlicer/OrcaSlicer #2394: FEATURE REQUEST - New prusaslicer 2.6.1 arrange objects system

insane78, 2023-10-11:
Please consider to get the new arrange objects on the bed from prusaslicer 2.6.1
It works like a charm.

Eldenroot, 2023-10-11:
I think it is in the progress :)

Noisyfox, 2023-10-12:
Its current implement has a huge limitation that it does not support multiple plates & wipe towers. I'm still working on improving it but it's quite complicated and still need more time.

Eldenroot, 2024-01-19:
Still relevant

Matszwe02, 2024-04-25:
I recently heard of a nice software that can arrange 3d models like prusa slicer 2.6, but supports multiple plates :

https://github.com/Rhoban/Plater

Maybe implementing that one's logic will be easier? They nicely explained it's workflow in Readme, and I think it's not so complicated

jeremytodd1, 2024-06-17:
I agree that Prusa's plate arrange system would be a very nice addition to Orca.

DDFranks, 2024-06-17:
prusa's nesting would be nice even if it was just for the single/per plate arrange. Instead of the global nesting arrange, maybe per plate would make more sense?

Matszwe02, 2024-09-18:
still relevant

> Orca bot: this issue is stale because it has been open for 90 days with no activity.

closed 2024-12-26 by github-actions[bot]