OrcaSlicer/OrcaSlicer #2395: Max volumetric speed per printer/profile

dderg, 2023-10-11:
**Is your feature request related to a problem? Please describe.**
I use different nozzles for the printer, they have different volumetric flow limits. I also use multiple different filament profiles, and the only way to have different volumetric flow limits is to double the amount of filament profiles that I have and always remember to pick the correct one. I'd rather have different nozzle profiles.

**Which printers will be beneficial to this feature**
Any printer that can switch nozzles

**Describe the solution you'd like**
To have volumetric flow limit in printer or process profile, like SuperSlicer has it.

**Describe alternatives you've considered**
Alternative solution is to use SuperSlicer. Or separate filament profiles like described above.

hatkyinc2, 2023-11-18:
@dderg Does the feature actually work for you, doesn't work for me and others https://github.com/SoftFever/OrcaSlicer/issues/2033
or did you just look for the settings?

wridgers, 2023-11-18:
I am also interested in this feature.

I think it's fine to keep "Max volumetric flow" as a filament property, but also add it as a printer property. The slicer could then take the minimum of the two. 0 to disable.

kvc0, 2023-11-23:
I think probably the place I'd expect to find this would be here, nearby nozzle size.

[image]

I agree with @wridgers that the right policy is `min(config.filament_max_volumetric_speed, config.nozzle_max_volumetric_speed)`, with 0 to disable the setting for each. This will let people configure filament maximums separately from their printer maximums. Some printers have higher flow capacity than others!

requies, 2024-01-10:
Agree it should be added.
I have "a bunch" of printers and they all basically differ per hotend/nozzle type.
Every printer has a different different max volumetric fllow.

(coming from superslicer i use this setting)

ardichoke, 2024-03-21:
I find it disappointing that @SoftFever opted to reject the above PR to address this issue. I only have two printers and already find it annoying having to duplicate filament profiles in order just to cap the volumetric max based on their different hotends, especially since OrcaSlicer doesn't have a way to lock filament profiles to specific printers.

requies, 2024-05-11:
Any news on this?

closed 2024-08-18 by github-actions[bot]