OrcaSlicer/OrcaSlicer #2398: bridges printed in mid air

dachu02, 2023-10-12:
### OrcaSlicer Version

1.7.0

### OS version

win11

### Printer

Bambu p1p

### How to reproduce

[Wooden_bridge-top.zip](https://github.com/SoftFever/OrcaSlicer/files/12885875/Wooden_bridge-top.zip)
attached model sliced in PrisaSlicer and Orca gives 2 different results (correct one in Prusa, wrong in Orca)

Brisges are printed in mid air and not connected to the perimeters, see picture below (left prusa, right orca)

I was trying to play with different settings but cant achieve results as in prusa.

### Actual results

[image]
on the right

### Expected results

[image]
on the left

### Project file & Debug log uploads

[Wooden_bridge-top (2).zip](https://github.com/SoftFever/OrcaSlicer/files/12885900/Wooden_bridge-top.2.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

igiannakas, 2023-10-25:
Try the latest nightly. It has a number of improvements in bridging and internal infill handling.

dachu02, 2023-10-25:
i'm using 17.0, is there a newer version?

igiannakas, 2023-10-25:
Yes, the nightly builds, 1.8 beta 1. You can find it on the actions tab above.

dachu02, 2023-10-25:
checking, thanks

dachu02, 2023-10-25:
[image]
works perfect, as expected :)
thanks guys

closed 2024-01-24 by dachu02