OrcaSlicer/OrcaSlicer #2399: Bambu X1C unnecessarily purges filament before every single print

discip, 2023-10-12:
I'm not entirely sure if this can be solved here, or if the behavior is backed into the **`BL X1C`** firmware.

So the issue I'm having (There are actually many more out there suffering from this!) is that the **`BL X1C`** does purge filament to remove residue from the previous filament to start a clean, color-wise homogeneous, print, which is perfectly fine **IF there was another filament used prior to the current print**.

However, this behavior is totally unnecessary if the same filament is used for the next print!
This is even irresponsible and wasteful as resources, money and most **critical**: time gets annihilated!

I thought about making **`BL X1C`** _remember_ which filament was used prior to an upcoming print and, based on that, purge filament or **NOT**.
What would make the whole AMS and extruder-setup live even longer would be to not wind the filament up after every print, just in case you want to use the same filament if you however choose to use different filament, do the rewind before the print starts!

Hopefully that makes sense! 😊

thanks in advance
kind regards

davidja92, 2023-10-12:
This won’t be a slicer issue but more of the start g code. If you can find the parameters that’s specify the previous filament or ams slot you should be able to use a if and then syntax to bypass the purge. I’m not sure if the printer stores the last filament used in a code we can call to

discip, 2023-10-12:
Thanks!
I understand that, but I thought someone in the know could tell which relevant code I need to add and which to change. 😅

discip, 2023-10-12:
Thanks!
I understand that, but I thought someone in the know could tell me which relevant code I need to add and which to change. 😅

davidja92, 2023-10-12:
I’ve messed with it I’ll see what I can dig up tonight.

Noisyfox, 2023-10-13:
Try something like this
https://www.printables.com/model/344496-fast-start-gcode-improved-quick-bed-leveling-end-r

davidja92, 2023-10-13:
> Try something like this https://www.printables.com/model/344496-fast-start-gcode-improved-quick-bed-leveling-end-r

Yea I have used it before it disables the retract and prime completely but has no logic to start the retract and purge if filament has changed

Noisyfox, 2023-10-13:
> > Try something like this https://www.printables.com/model/344496-fast-start-gcode-improved-quick-bed-leveling-end-r
> 
> Yea I have used it before it disables the retract and prime completely but has no logic to start the retract and purge if filament has changed

Yeah, you have to switch profile manually based on the previous printed filament.

discip, 2023-10-13:
> I don't think this can be fixed on the slicer side.

This is what I feared was true.

So the firmware of the BL X1C would need to be altered.

closed 2023-10-13 by discip

discip, 2023-10-13:
Btw:
I just remembered, that if you cancel a print the rewind is not performed and thus the next print is started without the extensive and totally unnecessary purge.

Is there a way to cancel a print via **`end g-code`**?