OrcaSlicer/OrcaSlicer #2406: Crashing at start

stiffconcepts, 2023-10-13:
### OrcaSlicer Version

V 1,7,0

### OS version

windows 10 

### Printer

Ratrig Vcore 3.1

### How to reproduce

No issues at first. I was doing PA calibration for new filament and slicer crashed. Tried to reopen slicer and received following errors and slicer wont load.
[image]
[image]

### Actual results

slicer fails to open. i have deleted and redownloaded several times even tired v 1.6.6 and same issue. 

### Expected results

need help.

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ocraftyone, 2023-10-13:
Please provide the crash dump logs

Ocraftyone, 2023-10-13:
And if you just want a semi quick fix. Just delete %appdata%/Orcaslicer. Be warned, **this will delete all custom settings!** It will be as if you just installed orca for the first time

closed 2024-01-20 by github-actions[bot]