OrcaSlicer/OrcaSlicer #2407: No horizontal seam

gogsi1, 2023-10-13:
**Which printers will be beneficial to this feature**
For everyone

**Describe the solution you'd like**
So that the seam is horizontal.

Seam on the edge: 
[image]

Seam at the back:
[image]

gogsi1, 2023-10-13:
> Did you mean **_vertical_** (perpendicular to the build-plate) instead of _horizontal_?
Yes, you got it right. I made a little mistake

discip, 2023-10-13:
Just to clarify:
So what you want is an absolutely vertical seam (90°) right?

igiannakas, 2023-10-25:
Try the latest nightly. It has the option to set a straight seem that is vertical to the build plane.

closed 2024-01-24 by gogsi1