OrcaSlicer/OrcaSlicer #2408: Pressure advance calibration uses printersettings-retract instead of filamentsettings-retraction

blackhead1981, 2023-10-13:
### OrcaSlicer Version

1.7.0-beta

### OS version

Win11

### Printer

Bambulab X1C

### How to reproduce

1. go to filament settings - setting overrides
2.  set a retractionlength **0.2mm** (in my case **0.8mm** is set under printer settings - extruder)
3. go to calibration - select pressure advance - DDE/PA pattern
4. slice
5. search in gcode for the retraction. 
You will see the is 0.8mm retraction instead of 0.2mm.
If you choose 0.2mm under printer settings it will be 0.2mm in gcode aswell.

### Actual results

You will see the is 0.8mm retraction instead of 0.2mm.
If you choose 0.2mm under printer settings it will be 0.2mm in gcode aswell.

### Expected results

In my opinion, the retraction length from the filament settings should be used when they are set

### Project file & Debug log uploads

[Pressure Advance Test.zip](https://github.com/SoftFever/OrcaSlicer/files/12899297/Pressure.Advance.Test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

R4toLoko, 2023-10-16:
I can replicate that.
`;retraction_length = 0.2 (printer)` and `;filament_retraction_length = 0.4` but I do not know which one is being called.
GCODE has a single `G1 E-.4 F3000` call and then `G1 E-.2 F3000` onwards .
If I change to PA line `G1 E-.2 F3000` isn't called once but `G1 E-.4 F3000`. Bug.

igiannakas, 2023-12-27:
Slim chance at least I'll fix that. The retraction is only there to clean up the pattern so any would do.

closed 2024-04-04 by github-actions[bot]