OrcaSlicer/OrcaSlicer #2414: Add ability to toggle gap filling

wight554, 2023-10-14:
**Is your feature request related to a problem? Please describe.**
When printing TPU, the proccess of gap filling creates small bulges that are being covered by next layer of solid infill later on and makes the top wall not smooth. I believe  reducing extra movements will improve this experience a lot.

**Which printers will be beneficial to this feature**
Any

**Describe the solution you'd like**
Port from PrusaSlicer https://github.com/prusa3d/PrusaSlicer/pull/5999

**Describe alternatives you've considered**
Printing slower with higher retractions, but not really the best option for TPU. I also tried settingn gap infill speed to 0 as per https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/can-i-turn-off-gap-filling-in-the-prusa-slicer/ but it didn't work

SoftFever, 2023-10-14:
Hi @wight554 
Orca does has this feature.
It allows you to specify the gap fill threshold, offering more granular control.
[image]

wight554, 2023-10-14:
> Hi @wight554 Orca does has this feature. It allows you to specify the gap fill threshold, offering more granular control. [image]

It's not applied to bottom and top layers as per description, I think prusa's one works for all. Not sure tho

SoftFever, 2023-10-15:
The description need to be updated.
It applies to all layers.

discip, 2023-10-17:
@wight554
So this issue is solved I assume, right?
Please don't forget to close. 😊

wight554, 2023-10-17:
> @wight554 So this issue is solved I assume, right? Please don't forget to close. 😊

Well, the issue doesn't exist, but maybe we should keep it for tracking description fix?
If not needed, I'll close it for sure

SoftFever, 2023-10-17:
Thanks, the description has been fixed.

wight554, 2023-10-17:
Thanks, closing the issue

closed 2023-10-17 by wight554