OrcaSlicer/OrcaSlicer #2418: Filament Shrinkage setting doesn't equally modify all parts on a 3MF model import

enly1, 2023-10-14:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Printer

Bambu Labs - P1S

### How to reproduce

Import multi-part 3MF File.
Modify Shrinkage on selected filament to 99%
Notice component parts of model don't scale equally, resulting in gaps between perimeters of previously touching parts.

### Actual results

Gaps created between parts

[image]

### Expected results

No Gaps

[image]

### Project file & Debug log uploads

[Dessicant_Tray_Drafted_Filament_Dryer.zip](https://github.com/SoftFever/OrcaSlicer/files/12906683/Dessicant_Tray_Drafted_Filament_Dryer.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

enly1, 2023-10-14:
Gap present between lower arc in pictures and hex fill.

psiberfunk, 2023-10-29:
@SoftFever , is xy shrinkage something other than XY scaling of objects ?

SoftFever, 2023-10-30:
It scales each region(2d) after slcing the model.
It should equal to XY scale the model in 3d

SoftFever, 2023-10-30:
With that said, filament shrinkage occurs on a per-project basis. 
Each object is scaled from its center point(it's acutually more like offseting). 
So, if you want to apply shrinkage to multiple objects collectively, please select these parts and then create an assembly for them.

psiberfunk, 2023-10-30:
> With that said, filament shrinkage occurs on a per-project basis. Each object is scaled from its center point(it's acutually more like offseting). So, if you want to apply shrinkage to multiple objects collectively, please select these parts and then create an assembly for them.

Shouldn’t that be done for the user automatically though ? I guess my point is that the user expectation is that it behaves like scaling , which does keep everything together when applied ?

closed 2024-02-12 by github-actions[bot]