OrcaSlicer/OrcaSlicer #2420: Locks up slicing temp tower

dtdionne, 2023-10-14:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Printer

Voron 0

### How to reproduce

1. Create temp tower
2. Choose .15 Optimal
3. Slice

All other layer heights slice without issue.  Here’s a video https://youtu.be/3awFJZD80nU?si=CZmHJqMlOnM6uqnI

### Actual results

https://youtu.be/3awFJZD80nU?si=CZmHJqMlOnM6uqnI

### Expected results

It should slice the temp tower

### Project file & Debug log uploads

[OrcaLogs.zip](https://github.com/SoftFever/OrcaSlicer/files/12907911/OrcaLogs.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

pciebiera, 2023-10-26:
As a workaround you can try to right click the object on the prepare tab, and choose 'Fix Model', after that it would slice for me.

Jax1er, 2023-11-04:
1.7.0 worked fine for me but just updated to 1.8.0 and its doing the exact same thing as you describe.  stuck on 15% and complete lockup.  Did you find anything that worked to correct this?

closed 2024-02-20 by github-actions[bot]