OrcaSlicer/OrcaSlicer #2422: Bridge layer

AlexVikingos, 2023-10-14:
[image]
[image]
Bridge layers are not connected or making 100 small pieces not in one layer. No possible to expend it.
Basically if have object with hole or more holes than bridge make for all individual but not connected together. I try all possible settings and can find way how to do it. Compare to superslicer, prusa slicer and ideamaker all slice in one piece

Noisyfox, 2023-10-15:
Could you share the project?

AlexVikingos, 2023-10-15:
[https://drive.google.com/file/d/1iggmyspcU3t0ZwSpYqJVQr1jgaf-Sfu-/view?usp=sharing](url)

Noisyfox, 2023-10-15:
[image]

I can also reproduce your issue using SuperSlicer. Did you change any specific settings? Or could you share your SuperSlicer's project?

Noisyfox, 2023-10-15:
Also, with changes from this PR https://github.com/SoftFever/OrcaSlicer/pull/2382 it generates one single complete bridge layer:
[image]

AlexVikingos, 2023-10-15:
[https://drive.google.com/file/d/1rjHdeu4T0Nf5lGUSDaWTav04YP5KAF2X/view?usp=sharing, https://drive.google.com/file/d/1wlbtcQuVp5I1bGiOgccBJkCpNMWgwmMH/view?usp=sharing](url)
my super slicer settings, the same with prusa slicer and similar with ideamaker.

Noisyfox, 2023-10-15:
Interesting.
BTW could you give the dev build of this PR https://github.com/SoftFever/OrcaSlicer/pull/2382 a try and see if it solves your problem?

closed 2024-01-22 by github-actions[bot]