OrcaSlicer/OrcaSlicer #2425: Undo history (like PS)

Eldenroot, 2023-10-15:
**Is your feature request related to a problem? Please describe.**
Add a undo/redo history like Prusa Slicer has

**Which printers will be beneficial to this feature**
Easier slicing

**Describe the solution you'd like**
Add a mew popup window? Check the screenshot.

**Describe alternatives you've considered**
---

**Additional context**
[image]

Noisyfox, 2023-12-19:
I do like this history window, but I'm not sure how to adapt that to current Orca's UI since 1) Orca has undo/redo buttons on the title bar; 2) We already have too many icons on the top toolbar and it's really hard to squeeze another two without making life harder for those with small screens.

Eldenroot, 2023-12-19:
Hmm, what about to display a popup window if you click on the undo/redo on the title bar with the right mouse click?

Noisyfox, 2023-12-19:
> Hmm, what about to display a popup window if you click on the undo/redo on the title bar with the right mouse click?

Yeah that's something I could try

Eldenroot, 2023-12-21:
I think this is the only way :) I am looking forward!

Eldenroot, 2024-03-21:
Still needed

closed 2024-09-26 by github-actions[bot]