OrcaSlicer/OrcaSlicer #2428: Enhance seam painting - add more options

Eldenroot, 2023-10-15:
**Is your feature request related to a problem? Please describe.**
Missing seam painting options for detailed customization.

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
Add more options for seam painting like PS has. Could be available via expert mode?

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
[image]

Eldenroot, 2023-12-17:
@Noisyfox - any plans to backport this from Prusa?

Noisyfox, 2023-12-19:
What you show in screenshot is the support painting, not seam painting.
[image]
Orca and PS have identical seam painting options (with slightly different wording).

[image]
And for support painting the only thing missing from Orca is the Automatic painting, which I've attempted to add a long time ago but gave up due to how different the codebase were at that time. Might give it another shot since Orca's core codebase is much much closer to PS now.

Eldenroot, 2023-12-19:
Srry, my mistake - the screenshot is not the correct one.

You are right. 

"Automatic painting of supports will be nice :)

Phardbean, 2024-01-20:
This is exciting. Cleaning up the supports for minis to avoid nightmares is sooooo easy with this tool. I need it so badly.

Eldenroot, 2024-01-25:
Yes, exactly :)

Noisyfox, 2024-01-26:
I just did another attempt (https://github.com/Noisyfox/OrcaSlicer/tree/dev/auto-support and https://github.com/Noisyfox/OrcaSlicer/tree/dev/auto-support-2) a few days ago, and it turned out to again, more difficult than I expected.

Phardbean, 2024-01-26:
You need to recruit help.🤓 wish I was capable.

Eldenroot, 2024-01-26:
> I just did another attempt (https://github.com/Noisyfox/OrcaSlicer/tree/dev/auto-support and https://github.com/Noisyfox/OrcaSlicer/tree/dev/auto-support-2) a few days ago, and it turned out to again, more difficult than I expected.

Keep going, you can do this ;)

Fingers crossed!

Eldenroot, 2024-04-26:
Staleness

closed 2024-08-02 by github-actions[bot]

eVenom, 2024-08-05:
is this coming down the line?