OrcaSlicer/OrcaSlicer #2429: HMS_0300_0200_0001_0003 The nozzle temperature is abnormal,the heater is over temperature.

AlexMa2011, 2023-10-15:
### OrcaSlicer Version

1.80-dev

### OS version

macOS 14.0

### Additional system information

CPU : Apple M1 Max
Memory : 64GB

### Printer

Bambu Lab P1S+AMS

### How to reproduce

使用0.6mm配置，不修改任何参数，切片，打印。

### Actual results

完成准备后，开始打印5-6秒后提示：HMS_0300_0200_0001_0003：喷嘴温度异常，加热器过温。
实际温度显示0度

### Expected results

正常打印

### Project file & Debug log uploads

[Max volumetric speed test.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12909397/Max.volumetric.speed.test.3mf.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/12909399/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

AlexMa2011, 2023-10-15:
用最新的BS，版本 1.7.7.89，是没有这个问题的

SoftFever, 2023-10-20:
看了生成的gcode，没有异常
我注意到你温度设置到了290度，可能因为pid的波动触发警报了。
不清楚你BS是不是也是这个温度
这个应该是机器硬件的问题

AlexMa2011, 2023-10-20:
刚才测试了一下，又没问题了，奇怪

closed 2023-10-20 by AlexMa2011