OrcaSlicer/OrcaSlicer #2436: OrcaSlicer and bambu studio

WimVDK, 2023-10-16:
So just ordered my X1C with AMS? Obviously it hasn't arrived yet lo, that be quick. In the mean time i'm looking around OS as i know it's more advanced than the native bambu studio (the same way i use Cura over Creality Slicer for my CR10 Max). I already signed in wiith my bambu account, set up the proper printer profiles but that leaves 1 question.

In order for the X1C to show up in OS, do i need bambu studio installed as well? Couldn't find that info in the OS documentation.

Ocraftyone, 2023-10-18:
no. there is the option to install the bambu network plugin that provides feature parity with bambu studio.

WimVDK, 2023-10-18:
I did install that one, Printer should arrive tomorrow. Then i'll be able to test in the upcoming days. But to clear just so there's no missunderstanding. OrcaSlicer is standalone and (although forked from BambuSlicer) does **NOT** require BambuSlicer to be installed?

Ocraftyone, 2023-10-18:
Yup! 100 percent correct!

Ocraftyone, 2023-10-18:
if that solves your issue, don't for get to close 😄

WimVDK, 2023-10-18:
if you don't mind i'll leave it open until i've confirmed this affter setting up the printer.

WimVDK, 2023-10-19:
We are up and running!

closed 2023-10-19 by WimVDK