OrcaSlicer/OrcaSlicer #2457: Add support for Prusa XL with up to five heads

rkarlsba, 2023-10-18:
**Is your feature request related to a problem? Please describe.**
Prusa XL is not supported so since PrusaSlicer can't read the extensions to 3mf added by Bambu and used by Orca, multicolour prints can't be moved between the two.

**Which printers will be beneficial to this feature**
Prusa XL

**Describe the solution you'd like**
Add Prusa XL

GuyH77, 2023-10-26:
I'd love to see this support added as well.

toddanglin, 2023-11-02:
BIG +1 to seeing support for Prusa XL (with tool changers) come to OrcaSlicer! I got spoiled to things OrcaSlicer has that PrusaSlicer lacks (like built-in calibration tests and support for brim ears). Would love to have those capabilities when working with my XL.

teachingtechYT, 2023-11-02:
Adding myself to this request. Back to using two slicers now that the XL has arrived. Really keen to only use OrcaSlicer.

freeformz, 2023-11-18:
Same. I'd like to standardize on OrcaSlicer over a mix.

freeformz, 2023-11-18:
Possbile dupe of #2050 and #2021

freeformz, 2023-11-18:
See also #2168 discussion.

tutulino, 2023-12-30:
It would be great to have this feature, this makes me use Orca for all my printers

clockworkcrow, 2024-02-02:
Indeed, XL suppport would be appreciated.

misisl, 2024-02-20:
Adding myself to this request. Back to using two slicers now that the XL has arrived. Really keen to only use OrcaSlicer.

jayjar291, 2024-02-26:
Bump would really want support for the prusa XL 2t

BrewNinja, 2024-03-04:
Would love this feature as well!

xidus90, 2024-03-08:
Also like to see this, special because the seam join was merged.

Bat-Maker, 2024-03-14:
+1....well 5, 5 Head version too please!

Hardy786, 2024-03-19:
+1 for Voron Stealth Changer

romass, 2024-03-19:
+1 for prusa XL

jan1409, 2024-04-03:
+1 here for Pruse XL - using Orca Slicer already for Bambu Printer and just received XL

matbee-eth, 2024-04-05:
+1 Vote for XL

dpark12300, 2024-04-24:
I have 2T XL.

Topless-Stang, 2024-05-10:
I have a 5T XL...was hoping it would be supported by now.  Hope it gets added soon.

fionnb, 2024-05-10:
me too! [tm]

kswallow54, 2024-05-11:
Ditto.. Would be great to use Orca over Prusa Slicer for the XL 5T

benjaminbrumbaugh, 2024-05-28:
Me too!
Prusa XL support might need to go through Bambu Studio first. Maybe we can start a gofundme to get SoftFever a 5-head XL :)

cochcoder, 2024-05-30:
Hey guys, I just made a PR (#5511) that should bring support for the Prusa XL with one print head. The only change I made from PrusaSlicer is having supports default to organic tree supports, rather than snug supports. I would love to hear your feedback and make sure that this printer works properly as I don't own it. I have a build running [here](https://github.com/cochcoder/OrcaSlicer/actions/runs/9309961649) and it should be finished within two hours of this comment. Is there a bug, does it work, do you have improvements that could be added? Please let me know in the PR!

hliebscher, 2024-05-31:
here is my working Start / End  code:
; 2024 Heiko Liebscher
; Printer : [printer_preset]
; Profile : [print_preset]
; Plate   : [plate_name]
; --- initial_extruder: [initial_extruder]
; --- has_wipe_tower: [has_wipe_tower]
; --- total_toolchanges: [total_toolchanges]
; --- T0: {is_extruder_used[0]}
; --- T1: {is_extruder_used[1]}
; --- T2: {is_extruder_used[2]}
; --- T3: {is_extruder_used[3]}
; --- T4: {is_extruder_used[4]}
;T[initial_extruder]
M17 ; enable steppers
M862.3 P "XL" ; printer model check
M862.5 P2 ; g-code level check
M862.6 P"Input shaper" ; FW feature check
M115 U5.1.2+13478
G90 ; use absolute coordinates
M83 ; extruder relative mode
; set print area
M555 X{first_layer_print_min[0]} Y{first_layer_print_min[1]} W{(first_layer_print_max[0]) - (first_layer_print_min[0])} H{(first_layer_print_max[1]) - (first_layer_print_min[1])}
; inform about nozzle diameter
{if (is_extruder_used[0])}M862.1 T0 P{nozzle_diameter[0]}{endif}
{if (is_extruder_used[1])}M862.1 T1 P{nozzle_diameter[1]}{endif}
{if (is_extruder_used[2])}M862.1 T2 P{nozzle_diameter[2]}{endif}
{if (is_extruder_used[3])}M862.1 T3 P{nozzle_diameter[3]}{endif}
{if (is_extruder_used[4])}M862.1 T4 P{nozzle_diameter[4]}{endif}

; turn off unused heaters
{if ! is_extruder_used[0]} M104 T0 S0 {endif}
{if ! is_extruder_used[1]} M104 T1 S0 {endif}
{if ! is_extruder_used[2]} M104 T2 S0 {endif}
{if ! is_extruder_used[3]} M104 T3 S0 {endif}
{if ! is_extruder_used[4]} M104 T4 S0 {endif}

M217 Z{max(zhop, 2.0)} ; set toolchange z hop to 2mm, or zhop variable from slicer if higher
; set bed and extruder temp for MBL
M140 S[first_layer_bed_temperature] ; set bed temp
G0 Z5 ; add Z clearance
M109 T{initial_tool} S{((filament_notes[initial_tool]=~/.*HT_MBL10.*/) ? (first_layer_temperature[initial_tool] - 10) : (filament_type[initial_tool] == "PC" or filament_type[initial_tool] == "PA") ? (first_layer_temperature[initial_tool] - 25) : (filament_type[initial_tool] == "FLEX") ? 210 : (filament_type[initial_tool]=~/.*PET.*/) ? 175 : 170)} ; wait for temp

; Home XY
G28 XY
; try picking tools used in print
G1 F{travel_speed * 60}
{if (is_extruder_used[0]) and (initial_tool != 0)}T0 S1 L0 D0{endif}
{if (is_extruder_used[1]) and (initial_tool != 1)}T1 S1 L0 D0{endif}
{if (is_extruder_used[2]) and (initial_tool != 2)}T2 S1 L0 D0{endif}
{if (is_extruder_used[3]) and (initial_tool != 3)}T3 S1 L0 D0{endif}
{if (is_extruder_used[4]) and (initial_tool != 4)}T4 S1 L0 D0{endif}
; select tool that will be used to home & MBL
T{initial_tool} S1 L0 D0
; home Z with MBL tool
M84 E ; turn off E motor
G28 Z
G0 Z5 ; add Z clearance

M104 T{initial_tool} S{if is_nil(temperature_vitrification[initial_tool])}70{else}{temperature_vitrification[initial_tool]-80}{endif} ; set idle temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
G29 G ; absorb heat

M109 T{initial_tool} S{((filament_notes[initial_tool]=~/.*HT_MBL10.*/) ? (first_layer_temperature[initial_tool] - 10) : (filament_type[initial_tool] == "PC" or filament_type[initial_tool] == "PA") ? (first_layer_temperature[initial_tool] - 25) : (filament_type[initial_tool] == "FLEX") ? 210 : (filament_type[initial_tool]=~/.*PET.*/) ? 175 : 170)} ; wait for temp

; move to the nozzle cleanup area
G1 X{(min(((((first_layer_print_min[0] + first_layer_print_max[0]) / 2) < ((print_bed_min[0] + print_bed_max[0]) / 2)) ? (((first_layer_print_min[1] - 7) < -2) ? 70 : (min(print_bed_max[0], first_layer_print_min[0] + 32) - 32)) : (((first_layer_print_min[1] - 7) < -2) ? 260 : (min(print_bed_max[0], first_layer_print_min[0] + 32) - 32))), first_layer_print_min[0])) + 32} Y{(min((first_layer_print_min[1] - 7), first_layer_print_min[1]))} Z{5} F{(travel_speed * 60)}
M302 S160 ; lower cold extrusion limit to 160C
G1 E{-(filament_type[0] == "FLEX" ? 4 : 2)} F2400 ; retraction for nozzle cleanup
; nozzle cleanup
[... 122 lines cut ...]
G0 X{(initial_tool == 0 ? 30 : (initial_tool == 1 ? 150 : (initial_tool == 2 ? 210 : 330))) + ((initial_tool == 0 or initial_tool == 2 ? 1 : -1) * 40) + ((initial_tool == 0 or initial_tool == 2 ? 1 : -1) * 3)} Z{0.05} F{8000} ; wipe, move close to the bed
G0 X{(initial_tool == 0 ? 30 : (initial_tool == 1 ? 150 : (initial_tool == 2 ? 210 : 330))) + ((initial_tool == 0 or initial_tool == 2 ? 1 : -1) * 40) + ((initial_tool == 0 or initial_tool == 2 ? 1 : -1) * 3 * 2)} Z0.2 F{8000} ; wipe, move quickly away from the bed
G1 E{- 1.5 * retraction_length[initial_tool]} F2400 ; retract
G92 E0 ; reset extruder position

M593 X T2 F35.8
M593 Y T2 F35.4

=====================================================End==============================
G4 ; wait
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+5, max_print_height)}{endif} ; Move bed down
P0 S1 ; park tool
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+97, max_print_height)} F300{endif} ; Move bed further down
; turn off extruder heaters
{if is_extruder_used[0]} M104 T0 S0 {endif}
{if is_extruder_used[1]} M104 T1 S0 {endif}
{if is_extruder_used[2]} M104 T2 S0 {endif}
{if is_extruder_used[3]} M104 T3 S0 {endif}
{if is_extruder_used[4]} M104 T4 S0 {endif}
M140 S0 ; turn off heatbed
M107 ; turn off fan
M221 S100 ; reset flow percentage
M84 ; disable motors
M77 ; stop print timer
; max_layer_z = [max_layer_z]

cochcoder, 2024-05-31:
> here is my working Start / End  code:
>
> ; 2024 Heiko Liebscher
>
> ; Printer : [printer_preset]
>
> ; Profile : [print_preset]
>
> ; Plate   : [plate_name]
>
> ; --- initial_extruder: [initial_extruder]
>
> ; --- [...]

Thanks for the G-Code, but this seems to be different than Prusa's. Does the current profile not work for you?

cochcoder, 2024-05-31:
Anyone wanting to test out the profile will need to use this new [build](https://github.com/cochcoder/OrcaSlicer/actions/runs/9324083950), as it configures the printer infill to be what is used in PrusaSlicer and fixes the filament start G-Code not working properly.

gltovar, 2024-06-01:
> Anyone wanting to test out the profile will need to use this new [build](https://github.com/cochcoder/OrcaSlicer/actions/runs/9324083950), as it configures the printer infill to be what is used in PrusaSlicer and fixes the filament start G-Code not working properly.

Are you able to write down a quick bit of what is needed to get going? I tried the windows portable version and I couldn't get to the point I could successfully slice anything with only the prusa xl selected as the printer.

cochcoder, 2024-06-01:
> Are you able to write down a quick bit of what is needed to get going? I tried the windows portable version and I couldn't get to the point I could successfully slice anything with only the prusa xl selected as the printer.

Did it throw a error when you tried to slice? One troubleshooting step you could try would be copying [these](https://github.com/user-attachments/files/15523482/Prusa.XL.zip) files into /path-to-portable/resources/profiles and overwriting any files when prompted. **Please do not do this if you have custom profiles in that portable**

FYI, there is the occasional bug where profiles won't show up until you close Orca Slicer and open it again

gltovar, 2024-06-01:
Thanks, that profile was helpful in getting me up to speed and slicing thanks!

cochcoder, 2024-06-01:
Let me know if everything works as expected and if there is any noticeable differences with Prusaslicer

gltovar, 2024-06-02:
I am an orca slicer newb so if you wanted to highlight an interesting option or two that are now opened to me I'll give'em a go. But i'll let you know if I notice any issues

cochcoder, 2024-06-02:
It would be interesting to see how scarf seems could be added, but I'd rather have a profile in Orca Slicer that preforms the same if not better than PrusaSlicer.

gltovar, 2024-06-03:
I get a bluescreen, though I just realized that a single head prusa XL is different from any of the tool changer prusa's as it is a fixed tool head (which makes sense).  I thought this might just only offer support for printing with the first tool (and I have the 5t). Sorry I couldn't be more help testing you change.

rkarlsba, 2024-06-03:
> I get a bluescreen, though I just realized that a single head prusa XL is different from any of the tool changer prusa's as it is a fixed tool head (which makes sense). I thought this might just [...]

Regardless, that really shouldn't trigger a bluescreen!

cochcoder, 2024-06-03:
That bluescreen issue is strange, we could try and use @hliebscher 's G-Code and see if that would work with 5 heads. 

Start G-Code: 

> ; 2024 Heiko Liebscher
> ; Printer : [printer_preset]
> ; Profile : [print_preset]
> ; Plate : [plate_name]
> ; --- initial_extruder: [initial_extruder]
> ; --- has_wipe_tower: [has_wipe_tower]
> ; --- [...]

End G-Code: 

> G4 ; wait
> {if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+5, max_print_height)}{endif} ; Move bed down
> P0 S1 ; park tool
> {if max_layer_z < max_print_height}G1 [...]

DaTTcz, 2024-06-04:
I look forward to being able to test it on my 5 heads. I am not satisfied with PrusaSlicer. So I'm very pleased that it will be in OrcaSlicer.

cochcoder, 2024-06-05:
I have a new build [here](https://github.com/SoftFever/OrcaSlicer/actions/runs/9375592240?pr=5511) that fully implements all the profiles and filaments for the Prusa XL with one head. Again, make sure to make backups of your existing configs just in case, especially if you use a installer. 

If someone gets a working start and end g-code for the Prusa XL with more than one print head, please reach out and let me know. As I don't own this printer, so I can't test adapting the G-Code from PrusaSlicer myself.

BrewNinja, 2024-06-05:
To get this straight, you just added a Prusa XL printer profile correct? So we are still waiting on multiple extruder support (which softfever said they would look at).

cochcoder, 2024-06-05:
> To get this straight, you just added a Prusa XL printer profile correct? So we are still waiting on multiple extruder support (which softfever said they would look at).

Yes that is correct. I only added the profile for one print head with the hopes of it working with up to five print heads (with only one actually printing) once the G-Code gets worked out. We are still waiting for multiple extruder support, one of the reasons I made this profile is to hopefully encourage someone to add support. As everyone working on this project is volunteering their time and it's not easy porting features from PrusaSlicer, especially when it's something as complex as multiple extruders.

DaTTcz, 2024-06-05:
> If someone gets a working start and end g-code for the Prusa XL with more than one print head, please reach out and let me know. As I don't own this printer, so I can't test adapting the G-Code from PrusaSlicer myself.

So far I've only ever used one print head (one to five), but I have no problem trying anything. I can slicer something and try to print with all five heads. Or do some testing.

cochcoder, 2024-06-05:
> So far I've only ever used one print head (one to five), but I have no problem trying anything. I can slicer something and try to print with all five heads. Or do some testing.

I'm a bit confused by your comment, sorry about that, are you saying you have both one and five print heads? If you have one can you do a test print to make sure the profiles work properly? Doesn't even need to finish. If you have more than one print head, does the current profile happen to work for you?

BrewNinja, 2024-06-05:
> Yes that is correct. I only added the profile for one print head with the hopes of it working with up to five print heads (with only one actually printing) once the G-Code gets worked out. We are [...]

Totally agree with the not easy porting features, and I hope I didn't come off as ungrateful to all the volunteers. I was just trying to clarify. Thanks!

DaTTcz, 2024-06-05:
> I'm a bit confused by your comment, sorry about that, are you saying you have both one and five print heads? If you have one can you do a test print to make sure the profiles work properly? Doesn't [...]

I apologize for my English, I speak Czech. I have a Prusa XL with five heads at work. So far I have only printed with each head separately, printing generated by PrusaSlicer. Tomorrow I will try the print generated by OrcaSlicer. Do I understand correctly that at this point you want to know if the printing will even start ?

cochcoder, 2024-06-05:
Yes, at this point I'm waiting to see if someone can get a print going with 2 or more heads.

GuyH77, 2024-06-05:
Love the ambition but I'd just wait for the port from Prusaslicer to Orca. Don't know why you'd try to do it all now via gcode when there's so much more in the slicer that impacts multi material print quality

cochcoder, 2024-06-05:
Yea, I admit, it is ambitious. I just figured I'd get into a state where once it's ported only a few changes are needed. And hey, having the convenience of doing simple prints in one slicer, no matter the version, would be a nice plus for now :)

rkarlsba, 2024-06-05:
> I'm a bit confused by your comment, sorry about that, are you saying you have both one and five print heads? If you have one can you do a test print to make sure the profiles work properly? Doesn't [...]

+1 for starting out with the basics instead of trying to swallow the whole thing at once

cochcoder, 2024-06-06:
Great, thank you so much for testing that! I'll go ahead and fix that as well as adding input shaper!

UmBsublime, 2024-06-08:
Worked fine for me as well. I had the same issue about output file name.
Checking out my other profiles replacing `{printing_filament_types}` with `{filament_type[initial_tool]}' fixed this for me.

cochcoder, 2024-06-08:
That issue has been fixed and the PR is now merged

Topless-Stang, 2024-06-10:
Has the printer profile been added, or do we need to use the custom g code?

cochcoder, 2024-06-10:
> Has the printer profile been added, or do we need to use the custom g code?

You won't need to use the custom G-Code, the profile will show up in the next release.

Topless-Stang, 2024-06-10:
Awesome, will wait for the next release.  awesome work!  By the way, I understand only one tool head will work, does it have to always be the first tool head thats used?

cochcoder, 2024-06-10:
> Awesome, will wait for the next release.  awesome work!  By the way, I understand only one tool head will work, does it have to always be the first tool head thats used?

Thanks! Yes only one head will work, I'd imagine that it'll be the first tool head as the start G-Code (from PrusaSlicer) is made for one head and there is no system for you to pick one in Orca.

DaTTcz, 2024-06-11:
If you want to print with a different head, you must upload the gcode into the printer and manually select a different head on the printer display.

benjaminbrumbaugh, 2024-06-12:
> @DaTTcz For some reason this is not the case for me. I just tested this and the menu to select tool-head never shows up and it simply defaults to tool-head 1

@umbsublime I haven’t tried this PR, but I do know that screen doesn’t show up normally if you’re using prusa connect and setting the printer ready remotely.

UmBsublime, 2024-06-12:
My test was done localy on the printer, te tool-head screen does not show up for me.

freeformz, 2024-06-13:
@cochcoder Thanks for bringing this one step closer to reality! Sorry I couldn't test as I only saw this now.

DaTTcz, 2024-06-14:
I don't know, I'll take a picture of the display menu on Monday. I can choose the head I want to print before each print.

DoubleStrike, 2024-06-25:
Not sure if it will help, but here is the latest version of the start and end gCode for a multi-head XL that also includes the recommended tweaks to nozzle temps during the bed meshing step to prevent oozing.

[end.txt](https://github.com/user-attachments/files/15979848/end.txt)
[start.txt](https://github.com/user-attachments/files/15979849/start.txt)

[4 more replies]