OrcaSlicer/OrcaSlicer #2458: Bridge: auto-direction-detection should not use the same direction as the infill above

Noisyfox, 2023-10-19:
The issue originated from https://github.com/SoftFever/OrcaSlicer/pull/2382#issuecomment-1761721677

In some cases the auto-detected bridge direction will be the same as the infill direction of the upper layer, which could cause quality issue if the bridge density is relative low.

We could consider improving this by adding some sort of constraint to the algorithm so it avoids using the same direction as the infill direction as much as possible.

SoftFever, 2023-10-20:
Are you referring to these highlighted areas?
[image]

Noisyfox, 2023-10-20:
> Are you referring to these highlighted areas? [image]

Nope, the other part.

SoftFever, 2023-10-20:
Damn, not sure why I circled that parts. Need more coffee..

closed 2024-01-27 by github-actions[bot]