OrcaSlicer/OrcaSlicer #2466: what is this? the little piece of extraction beside warped wall

wonsuhna1, 2023-10-20:
### OrcaSlicer Version

1.8.0.-dev (nightley)

### OS version

windows 10

### Printer

diy printer.

### How to reproduce

what is this? the little piece of extraction beside warped wall
in my case, It happened almost every time. This also happened in version 1.7.0.

### Actual results

[image]

### Expected results

There should be no meaningless halucination extrusions.

### Project file & Debug log uploads

[turbo cooler mount v1.zip](https://github.com/SoftFever/OrcaSlicer/files/13049199/turbo.cooler.mount.v1.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2023-10-20:
Have you tried the latest master build? I think this has been fixed by the new ensure vertical thickness algorithm.

wonsuhna1, 2023-10-20:
> Have you tried the latest master build? I think this has been fixed by the new ensure vertical thickness algorithm.
> 
> [image]

I checked and it did indeed work.
Thanks

I'll close this post.

closed 2023-10-20 by wonsuhna1