OrcaSlicer/OrcaSlicer #2467: Can't add any AnyCubic printer from Printer Selection

mcrimaldi, 2023-10-20:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Printer

Anycubic Mega-S with Klipper firmware

### How to reproduce

Go to "Printer"
Add Printer
Select any AnyCubic printer to add

### Actual results

No printer added from AnyCubic presets. Any other printer is correctly added

### Expected results

The anycubic printers should be added

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13050897/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

mcrimaldi, 2023-10-20:
Looks like the problem is here (from another log):

```
2023-10-20 12:59:49.517012[Thread 0x000016b8]:Slic3r::PresetBundle::load_presets enter, substitution_rule 4, preferred printer_model_id Anycubic i3 Mega S
2023-10-20 12:59:49.519010[Thread 0x000016b8]:Slic3r::PresetBundle::load_vendor_configs_from_json: parse C:\Users\Mariano\AppData\Roaming\OrcaSlicer\system/Anycubic.json, got description:  "Anycubic configurations"
2023-10-20 12:59:49.562012[Thread 0x000016b8]:Slic3r::ConfigBase::load_from_json: parse C:\Users\Mariano\AppData\Roaming\OrcaSlicer\system/Anycubic/process/0.15mm Optimal @Anycubic Kobra2.json got a generic exception, reason = Invalid value provided for parameter sparse_infill_pattern: rectilinear
2023-10-20 12:59:49.562012[Thread 0x000016b8]:Slic3r::PresetBundle::load_vendor_configs_from_json::<lambda_4760ed3e8311b28387eaf1ce8ddb8f34>::operator (): load config file C:\Users\Mariano\AppData\Roaming\OrcaSlicer\system/Anycubic/process/0.15mm Optimal @Anycubic Kobra2.json Failed!
2023-10-20 12:59:49.562012[Thread 0x000016b8]:Slic3r::PresetBundle::load_vendor_configs_from_json, got error when parse process setting from C:\Users\Mariano\AppData\Roaming\OrcaSlicer\system/Anycubic/process/0.15mm Optimal @Anycubic Kobra2.json
2023-10-20 12:59:49.563011[Thread 0x000016b8]:Slic3r::PresetBundle::load_vendor_configs_from_json: parse C:\Users\Mariano\AppData\Roaming\OrcaSlicer\system/Custom.json, got description:  "My configurations"
```

mcrimaldi, 2023-10-20:
> Looks like the problem is here (from another log):
>
> ```
> 2023-10-20 12:59:49.517012[Thread 0x000016b8]:Slic3r::PresetBundle::load_presets enter, substitution_rule 4, preferred printer_model_id [...]

Changing this line in `\system/Anycubic/process/0.15mm Optimal @Anycubic Kobra2.json` solves the issue:

`"sparse_infill_pattern": "rectilinear",` to `"sparse_infill_pattern": "grid",`

OutlawNL, 2024-01-13:
Had the same issue on MacOS (file is in Macintosh HD/Users/<username>/Library/Application Support/OrcaSlicer/system/Anycubic/process)

This doesn't seem to be resolved in the latest version. Thanks, mcrimaldi, for the work-around.

closed 2024-01-15 by SoftFever

mb-software, 2024-02-09:
I am still having this issue even with the latest nightly build version. (Using the Linux AppImage). Shouldn't the fix be merged to that?

KAPTOHKA, 2024-02-21:
OrcaSlicer _**v1.9.1**_ & **_v2.0.0dev_**.
Windows 11 Pro.
No printers added from ```AnyCubic``` presets. 
Any other printer is correctly added...
[Win11_logs.zip](https://github.com/SoftFever/OrcaSlicer/files/14360988/Win11_logs.zip)

navi21, 2024-11-24:
Cant add any printer, tried adding bambu but after selecting bambu labs it says no vendor selected and model drop down does not work