OrcaSlicer/OrcaSlicer #2468: Orcaslicer makes support appear on top of item, not towards build plate.

andreaselme, 2023-10-20:
### OrcaSlicer Version

1.7.0

### OS version

Win11

### Printer

Bambu P1S

### How to reproduce

https://www.printables.com/model/2770-human-skull-anatomically-correct/files

Trying to print this model. I selected 0.16mm layer, enable tree support, enable variable layer height. After enabling variable layer height it will add tree structure ON TOP of the model i.e. not towards build plate. 

### Actual results

 Support appears through the object, it seems. And on top of it.
[[https://i.postimg.cc/GtcRFx3m/Screenshot-2023-10-20-114627.png](url)](https://i.postimg.cc/GtcRFx3m/Screenshot-2023-10-20-114627.png)

### Expected results

I would expect normal tree supports.

### Project file & Debug log uploads

https://i.postimg.cc/GtcRFx3m/Screenshot-2023-10-20-114627.png

### Checklist of files to include

- [ ] Log file
- [X] Project file

RickTee, 2023-10-21:
Same X1C only tried with tree supports. Preview shows supports over the top of the model supporting nothing, print comes out the same.
[image]

marksventsitsky, 2023-10-21:
**OrcaSlicer Version**
1.7.0
**OS version**
MacOS

Similar problem when using automatic variable layer heights and using tree supports
[image]

Noisyfox, 2023-10-22:
Do not use default tree or organic tree + variable layer height, they are not compatible with each other. Next version will show an error when this happens.

closed 2024-01-29 by github-actions[bot]