OrcaSlicer/OrcaSlicer #2472: Issue with the new camera update

Noisyfox, 2023-10-20:
Caused by 388b48377434a6ee1a9ed8f6329c07a18d1a2157 

[image]

SoftFever, 2023-10-20:
I will take a look

discip, 2023-10-20:
Good evening everyone.
I'm experiencing this weird behavior quite for a while now.

@Noisyfox
Could you please try if this behavior is also caused by zooming in and out on any e.g. corner of your build plate **(zoom to mouse position enabled)**?
This is enough to trigger it for me.

If you than press **`Ctrl+0`** everything is back to normal.

discip, 2023-10-20:
Good evening everyone.
I'm experiencing this weird behavior quite for a while now.

@Noisyfox
Could you please try if this behavior is also caused by zooming in and out on any e.g. corner of your build plate **(zoom to mouse position enabled)**?
This is enough to trigger it for me and #2473 does unfortunately not solve it. 🤷🏻‍♂️

If you than press **`Ctrl+0`** everything is back to normal.

discip, 2023-10-20:
Good evening everyone.
I'm experiencing this weird behavior quite for a while now.

@Noisyfox
Could you please try if this behavior is also caused by zooming in and out on any e.g. corner of your build plate **(zoom to mouse position enabled)**?
This is enough to trigger it for me and #2473 unfortunately doesn't solve it. 🤷🏻‍♂️

If you than press **`Ctrl+0`** everything is back to normal.

Noisyfox, 2023-10-21:
> Good evening everyone. I'm experiencing this weird behavior quite for a while now.
>
> @Noisyfox Could you please try if this behavior is also caused by zooming in and out on any e.g. corner of your [...]

I can trigger the issue with your method with the master build, and with the https://github.com/SoftFever/OrcaSlicer/pull/2473 I could no longer reproduce it. The PR contains 2 commits, did you test with both commits?

discip, 2023-10-21:
@Noisyfox
Ah, yes you are right!
That fixed it for me!
Tried this to see if that would solve the weird behavior prior to the #2473!

Thanks!
kind regards

PR #2473 (2023-10-21): More camera tweaks
fixes #2472
closed 2023-10-21 by SoftFever