OrcaSlicer/OrcaSlicer #2474: Orcaslicer Gcode Crashes Voron 2.4 MCU

s3anami, 2023-10-21:
### OrcaSlicer Version

1.7.0

### OS version

Windows 11 Pro

### Additional system information

Memory: 64GB
CPU: AMD 5950X
Display Card: 3080 Ti

### Printer

Voron 2.4

### How to reproduce

Anything I print gcode generated with Orcaslicer, it crashes the MCU of the Voron.
This happens about 2-3% into the print.
I have generated gcode in PrusaSlicer and SuperSlicer and do not experience this issue.
I have tried what some have suggested and started increasing the gcode_arcs value to 0.2 and 0.3 in my printer.cfg and nothing changes. MCU still crashes.

### Actual results

Anything I print gcode generated with Orcaslicer, it crashes the MCU of the Voron.
This happens about 2-3% into the print.
I have generated gcode in PrusaSlicer and SuperSlicer and do not experience this issue.
I have tried what some have suggested and started increasing the gcode_arcs value to 0.2 and 0.3 in my printer.cfg and nothing changes. MCU still crashes.

### Expected results

Print should not crash MCU

### Project file & Debug log uploads

[debug_network_Fri_Oct_06_21_36_25.log](https://github.com/SoftFever/OrcaSlicer/files/13059859/debug_network_Fri_Oct_06_21_36_25.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

R4toLoko, 2023-10-24:
Klipper will have a log explaining why it is failing.
I have LDO Trident and have no had a single crash because of the slicer.
Although I have no idea of what this does, mine looks like:

```
# Enable arcs support
[gcode_arcs]
resolution: 0.1
```

Also, many use Raspberry Pi with their Voron which has restricted hardware performance imo. I use a i5 thin client (miniPC).
Some features on Orcaslicer might be too much for the Pi you are using crashing the printer MCU. Wild guess.

I have been using OrcaSlicer former SoftFever since version 0.0.1 and this is the first time I see a Voron having this problem because of it.

s3anami, 2023-11-05:
right it was a suggestion to fix the overloading to raise the resolution, it does nothing though

dawm, 2023-11-19:
Having the same issue, I've got 2 identical failed prints which died at the exact same spot/layer/time and its also during a G3 arc command.
[image]
[image]

closed 2024-02-25 by github-actions[bot]