OrcaSlicer/OrcaSlicer #2477: Off for and full speed at settings for auxiliary fan

mard89, 2023-10-21:
I have a rather powerful auxiliary fan that has the potential to, at lower layer counts, cool the print bed. Causing issues with print separating from the bed.

Currently orcaslicer powers on the auxiliary fan at the first layer. I would prefer the same control over the auxiliary fan as I do over the regular part cooling fan.

Stevetm2, 2023-10-22:
If you are using klipper you can make the aux fan use the same speed (or a scaled value) as the part fan...  this is just a work around of course, if you have klipper...  my aux fan is set as back_fan btw...

[gcode_macro M106]
description: Fan speed + back fan speed
rename_existing: M99106
gcode:
    M99106 {rawparams}
    {% set s = params.S|float %}
    SET_FAN_SPEED FAN=back_fan SPEED={s/255.0}

mard89, 2023-10-23:
> If you are using klipper you can make the aux fan use the same speed (or a scaled value) as the part fan... this is just a work around of course

I actually have a similar setup at the moment in klipper. The 2 problems I'm running into is that I don't want the run the auxiliary fan when the main part cooling fan is sufficient (for instance when printing ABS), and I would like to  set a different height at which the fan is set to it's max duty cycle.

Hence the request to be able to control the ramp up of the auxiliary fan in a similar fashion as the part cooling fan.

rushl, 2023-11-17:
I don't have Klipper, but would still like to set the layer to turn on the Aux fan. It seems like a good general parameter to have.

Paradisebaker, 2024-01-27:
I have the same kind of workaround with controlling the aux fan via the main parts fan, but this can backfire on you depending on the scenario. It would be nice to have the same set of controls for the aux fan as for the parts cooling fan, including the layer time, bridge, overhangs etc. options.

closed 2024-05-05 by github-actions[bot]