OrcaSlicer/OrcaSlicer #2480: RepRapFirmware M205 and M204 not populating in output Gcode (Jerk and acceleration)

Construct3D-AM, 2023-10-21:
### OrcaSlicer Version

1.7.0

### OS version

Win 10

### Additional system information

Ryzen 7950X
GTX 1080
RTX 3090
64GB DDR5 RAM

### Printer

Construct 1

### How to reproduce

Creating a Gcode file sets the initial jerk to X5 and Y5
Additionally, travel moves are also set to 1500mm/s^2 instead of what is set inside the slicer.
[image]
[image]

This is repeatable every time a Gcode file is made. Appears to be a bug similar/identical to **#1505**

### Actual results

M205 X5 Y5
M204 P4000 T1500

### Expected results

M205 X10 Y10
M204 P4000 T1500

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13061938/log.zip)
[test shape.zip](https://github.com/SoftFever/OrcaSlicer/files/13061940/test.shape.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Construct3D-AM, 2023-10-22:
Update:

When changing the G-Code firmware flavour to RRF. the motion ability tab is removed.
By changing the printers G-code flavour to Marlin. I can see under the now visible motion ability tab that the printer was set to jerk values of 5X and 5Y which lines up with the G-code file.

I would hazard a guess that the real issue isnt the Jerk values improperly appearing in the output file, but instead the motion ability tab being missing so users do not understand why jerk values of 5 are being used.

Construct3D-AM, 2023-10-22:
In regards to the acceleration M204 oddity.
It appears that the Marlin2 feature for Maximum Acceleration for travel is getting inserted into the G-code output
[image]
[image]

igorscheffer, 2023-11-24:
I had a similar problem, I discovered it was the printer settings. for example, the maximum jerk was set to 9, so when I put something higher in the filament configuration it wouldn't