OrcaSlicer/OrcaSlicer #2482: Cannot see home screen text and some menu text on Linux

nwpulverizer, 2023-10-22:
### OrcaSlicer Version

1.7.0

### OS version

Arch Linux - Qtile WM

### Additional system information

CPU: i7 8700K
Memory: 32GB ddr4
GPU: RTX 3070

### Printer

Bambu P1s

### How to reproduce

1. Install orca slicer on Arch Linux. I am using Qtile and my window manager, but trying on others such as lxqt does not change the effect. 
2. Open Orca slicer
3. View home page or 
4. Press the prepare tab 
5. Click gear icon at the top next to printer

### Actual results

No text is visible. Mousing over items indicates that they are there, but there is something going on with the actual text. 
[image]
[image]

### Expected results

Be able to read text on home screen or printer screen. Cannot add new printers because of this. 

### Project file & Debug log uploads

[debug_Sat_Oct_21_17_22_31_15293.log.txt](https://github.com/SoftFever/OrcaSlicer/files/13062472/debug_Sat_Oct_21_17_22_31_15293.log.txt)
[Untitled.3mf.gz](https://github.com/SoftFever/OrcaSlicer/files/13062473/Untitled.3mf.gz)

### Checklist of files to include

- [X] Log file
- [X] Project file

nwpulverizer, 2023-10-31:
potentially related to #1344, #883

TexZeTech, 2023-10-31:
I do have this problem on my main system running Nvidia proprietary drivers but not on my laptop which doesn't have a dedicated GPU and is most likely running video-linux. They are both running Manjaro.

I'm really tempted to blow away my system and see if that would fix it but more and more I'm suspecting something with the Nvidia drivers as darco415 said they tryied out Nouveua drivers in https://github.com/SoftFever/OrcaSlicer/issues/920#issuecomment-1767570850

Desktop:
```
Graphics:
  Device-1: NVIDIA GP104 [GeForce GTX 1080] vendor: ASUSTeK driver: nvidia
    v: 535.113.01 alternate: nouveau,nvidia_drm non-free: 535.xx+
    status: current (as of 2023-09) arch: Pascal code: GP10x
    process: TSMC 16nm built: 2016-21 pcie: gen: 3 speed: 8 GT/s lanes: 8
    link-max: lanes: 16 bus-ID: 01:00.0 chip-ID: 10de:1b80 class-ID: 0300
  Display: x11 server: X.Org v: 21.1.8 with: Xwayland v: 23.2.1
    compositor: kwin_x11 driver: X: loaded: nvidia gpu: nvidia,blackmagic-io
    display-ID: :0 screens: 1
```
[Desktop_debug_Tue_Oct_31_17_26_41_41099.log](https://github.com/SoftFever/OrcaSlicer/files/13222011/Desktop_debug_Tue_Oct_31_17_26_41_41099.log)

I also used the command `journalctl -n 1500 > Desktop_journalctl.txt` 

[Desktop_journalctl.txt](https://github.com/SoftFever/OrcaSlicer/files/13222074/Desktop_journalctl.txt)

Laptop:
```
  Display: x11 server: X.Org v: 21.1.8 compositor: kwin_x11 driver: X:
    loaded: modesetting alternate: fbdev,vesa dri: crocus gpu: i915
    display-ID: :0 screens: 1
```

nwpulverizer, 2023-11-01:
@TexZeTech running orca from the terminal I can see there are some nvidia errors when opening the menus, so I can corroborate that hypothesis. A fix mentioned in #883 regarding GTK themes worked for me but it seems like a bandaid. 

`GTK_THEME=Light orca-slicer `

allowed me to access the menus.

closed 2024-02-12 by github-actions[bot]

lzhecz, 2024-03-17:
I still had this issue, fixed with next solution
add `env WEBKIT_DISABLE_COMPOSITING_MODE=1 ` to Exec in your Orca's Desktop Entry like this:

> [Desktop Entry]
> Name=OrcaSlicer
> Exec=**env WEBKIT_DISABLE_COMPOSITING_MODE=1** /opt/orca-slicer/bin/orca-slicer %F
> Icon=OrcaSlicer
> Type=Application
> Categories=Utility;
> MimeType=model/stl;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;application/x-amf;

(mine Desktop Entry located at  /usr/share/applications/OrcaSlicer.desktop)

before: 
[image]
after:
[image]

BenBergman, 2024-04-16:
I've got this same issue. `GTK_THEME=Light orca-slicer` works for me as a band-aid, but not `env WEBKIT_DISABLE_DMABUF_RENDERER=1`

TexZeTech, 2024-04-18:
> I've got this same issue. `GTK_THEME=Light orca-slicer` works for me as a band-aid, but not `env WEBKIT_DISABLE_DMABUF_RENDERER=1`

If you are running Nvidia hardware this might work for you.

https://github.com/SoftFever/OrcaSlicer/issues/920#issuecomment-2025895738

BenBergman, 2024-04-20:
I'm on AMD