OrcaSlicer/OrcaSlicer #2485: Klipper  Fan Control M106 does not support P=Fan Number

ALM007cs, 2023-10-22:
### OrcaSlicer Version

1.7.0

### OS version

Window 10

### Printer

Klipper general

### How to reproduce

Klipper does not support the fan number in M106 P=FanNumber s=FanSpeed.

### Actual results

the hotend fan is the target .........

### Expected results

When setting to Klipper Firmware ist orca should use SET_FAN_SPEED command

### Project file & Debug log uploads

[Abdeckung230V.zip](https://github.com/SoftFever/OrcaSlicer/files/13062875/Abdeckung230V.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SoftFever, 2023-10-23:
Hi
It is because you have activated the auxiliary fan which will issue `M106 P2`
Disable auxiliary fan in Printer settings if you don't have it.
-------continue reading if you have a auxiliary cooling fan and want to use it-------------
Since Klipper don't have standard command to support auxiliary fan, Orca follows Marlin standard.
You can override M106 command to support it though in Klipper.
I will update WIKI when releasing 1.8 as I have extended more on this in 1.8 to support exhaust fan too

closed 2023-10-29 by SoftFever

ALM007cs, 2023-11-01:
After the update to 1.8, an exhaust fan works with the following macro.
other fans can also be inserted...

[gcode_macro M106]
rename_existing: M106.1
gcode:
  {% if params.P is defined and params.P|int > 0 %}
    {% if params.P|int == 1 %}
       RESPOND MSG="Fan does not exist!"
    {% endif %}
    {% if params.P|int == 2 %}
       RESPOND MSG="Fan does not exist!"
    {% endif %}
    {% if params.P|int == 3 %}
       SET_FAN_SPEED FAN=exhaust_fan SPEED={(params.S|int / 255 )}  
    {% endif %}
  {% else %}
    M106.1 S{params.S|int}
  {% endif %}