OrcaSlicer/OrcaSlicer #2489: Print job select the wrong slot

jeor112, 2023-10-22:
### OrcaSlicer Version

1.7.0

### OS version

Windows 11

### Printer

BL X1C

### How to reproduce

1. All AMS slots loaded with filament and slot 1 and 2 is loaded white PETG Filament
2. Slicer synced with AMS
3. Select slot 2 in Process, Objects
4. Select Print plate

### Actual results

In the print dialog box it choose slot 1

### Expected results

When open print dialog box, I expect slot 2 as I have selected in the sllocer Process object.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jeor112, 2023-10-22:
Well, I have now experienced the same behaivor in the latest Bambu slicer version.
I have also noticed, that I have to sync/resync with the AMS two or three times before the slicer is updated with the correct filament type.
So, it's apparently not only a OrcaSlicer problem.

Rat-N, 2023-11-23:
Same issue here. Select a filament on the object on the build plate - filament is in AMS slot 1. When sliced there's no way to see which filament is chosen. When Print Plate selected, dialog shows slot 3 selected. Have to change to slot 1 before I hit print.

Rat-N, 2023-11-23:
To my mind, the whole AMS filament setup on the Prepare page is messy and confusing. Really needs to have specific 'fixed' external and AMS slots into which the filament is sync/resynced to or manual edit. The slots don't 'come and go' depending upon what's loaded. The 4/8/12/16 AMS slots are fixed placement and not addable/deletable. You assign the filament profile to the slot. If you have a generic filament in AMS slot 4 then that's where it should remain and not move to a lower slot when one of the lower slot filaments runs out.

closed 2024-03-01 by github-actions[bot]