OrcaSlicer/OrcaSlicer #2490: Problem Network-Plug-in OrcaSlicer in VM Win 7 on OSX

NonBlackstation, 2023-10-22:
I have Bambu Lab Studio on my laptop (virtual Machine with Win 7 64bit) and the network plug-in works there. The download also works and no other software accesses the files.
Now I have copied the folder under C:/........./AppData/Roaming/Bambu../plugins into the OrcaSlicer folder. However, this is not recognized and a download via OrcaS is not possible.
When I do the network test in Bambu Studio, it's OK. But with Orca the BBL page doesn't work and the http
[image]
[image]

NonBlackstation, 2023-10-24:
Is the problem perhaps because I am using the OrcaSlicer in a virtual machine with Windows 7 (Parallels)?
However, all programs and internet access work. The test works with BambuStudio itself.
The virtual machine has its own IP (bridged network).
On my wife's laptop (Win7 Pro 64bit) it works with the OrcaSlicer network plug-in.

SoftFever, 2023-10-29:
TBH, I never tested Win 7. It’s very likely not supported

NonBlackstation, 2023-10-29:
jippie, OrcaSlicer 1.8.0 beta 1 work fine on Win 7 Pro 64bit. No problem to download the Network PlugIn...  :) :)

discip, 2023-10-30:
@NonBlackstation
If this is fixed, than please don't forget to close. 😊

closed 2023-11-01 by NonBlackstation