OrcaSlicer/OrcaSlicer #2492: top layer not sliced with multi material, manual filament changes.

netweaver1970, 2023-10-22:
### OrcaSlicer Version

latest latest main, 1.8-Dev, 'Optimize some models' commit 12eafd5 (https://github.com/SoftFever/OrcaSlicer/commit/12eafd5491065b5e6e832840b930d94fb40a78c1)

### OS version

MacOS 12

### Additional system information

Apple MacBookPro M1 

### Printer

Voron 2.4, single extruder, no AMS

### How to reproduce

see attached project file. Load and slice, look at the preview

### Actual results

The print has 3 parts on top of each other, to be printed in 1 go but with 2 manual filament changes.
The bottom and top parts are supposed to be the negative text, black colour filament and only 1 layer high. The middle part is solid yellow, kind of the background for the text
Bad result: the top text layer (black) is not sliced in the result. Only bottom text layer and middle solid part are sliced.
The little preview at the top left is correct though, you can see the black top layer fine. But the actual sliced preview lacks the top layer.
All layers (apart from the bottom, which is 0.3mm) are 0.2mm. Printer is set to Single extruder with manual filament changes.
[image]

### Expected results

all 3 (not only bottom 2) parts sliced and shown in the preview. In that case also all 3 parts will be printed too, with 2 manual changes, not the 1 change as currently presented.

### Project file & Debug log uploads

[luggage Geert_proj.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13064621/luggage.Geert_proj.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SoftFever, 2023-10-30:
The issue is cuased you have set initial layer height to 0.3mm while you have 0.2mm high top/bottom parts.
Changing the initial layer height to 0.2 will fix your issue.

Also, you might want to add PAUSE/M600 command in change filament gcode so that it will pause the print properly.

netweaver1970, 2023-10-30:
thanks for looking into it. But I still think it's weird. The bottom layer (0.3mm) should only be for the layer touching the build plate. But it should not impact any other layer. The top object should be sliced independently. Or not?
The top object is 0.2mm tall, was it not sliced because the resulting height is now only 0.1mm (due to bottom layer being 0.3mm, taller than what was initially foreseen) ? 

Would variable layer height be able to fix this automatically, reduce the middle object?

Another option, if I made the bottom object also 0.3mm tall, then then top object could stay 0.2mm? Is this how it could work?

In the end I just used 1 big object and put in 2 manual pauzes in the preview screen at t he correct height, so I could swap filaments but It would be nice if I could just use independent objects and do the the filament assignment in OrcaSlicer. Also in light of future MMU addition.

In further attempts I did put the M600 in OrcaSlicer + proper macro in Klipper.

Thanks

closed 2024-02-12 by github-actions[bot]