OrcaSlicer/OrcaSlicer #2493: Unrecognized keys from Bambu Studio 1.7.7.89 3mf

bani6809, 2023-10-23:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Printer

Bambu X1C+AMS

### How to reproduce

Save 3mf from Bambu Studio. Open it in OrcaSlicer 1.7.0.

### Actual results

[image]

### Expected results

OrcaSlicer should support these new keys 🙂

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2023-10-25:
Latest nightly build (can be found in the actions section above) supports the latest Bambu slicer version. Give it a try.

SoftFever, 2023-10-29:
Fixed in 1.8 beta

closed 2023-10-29 by SoftFever