OrcaSlicer/OrcaSlicer #2494: Makerworld doesn't accept print profiles from OrcaSlicer

bani6809, 2023-10-23:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Printer

Bambu X1C+AMS

### How to reproduce

Go to makerworld. Select a model, then select add print profile. Select the 3mf generated from OrcaSlicer.

### Actual results

[image]

### Expected results

It should accept the print profile.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2023-10-25:
It can’t - orca slicer has many more options that Bambu studio doesn’t support. And since the slicing is happening in the “cloud” using Bambu slicer using an orca profile would be incompatible and would not allow the model to slice.

Vanilla009, 2023-10-27:
This is related to my issue #2223

SoftFever, 2023-10-29:
It’s up to Bambu to accept OrcaSlicer’s profile.
I’d suggest asking for Orca support in their site

psiberfunk, 2023-10-29:
This isn’t a valid bug for this repo.. it should be closed

closed 2024-02-12 by github-actions[bot]