OrcaSlicer/OrcaSlicer #2495: Seam position "Back" - Bambu Studio seams are near perfectly straight, OrcaSlicer 1.7.0 seams are wildly askew

bani6809, 2023-10-23:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Printer

Bambu X1C+AMS

### How to reproduce

Load attached 3mf, slice it in Orca Slicer 1.7.0 and Bambu Studio 1.7.7.89

### Actual results

Orca Slicer 1.7.0
[image]

### Expected results

Bambu Studio 1.7.7.89
[image]

### Project file & Debug log uploads

[devo energy dome 0.10mm.zip](https://github.com/SoftFever/OrcaSlicer/files/13065031/devo.energy.dome.0.10mm.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

faith-healer, 2023-10-26:
+1 (Whip It!)

SoftFever, 2023-10-27:
From what I recall, whether you get a straight seam or not in both slicers depends on the model and settings; it's not guaranteed.
Note that BambuStudio uses the classic wall generator by default, while Orca utilizes Arache. There are other differences as well.

That said, I recognize the preference many have for straight seams. So, I've added a feature that lets users draw a perfectly straight seam line: https://github.com/SoftFever/OrcaSlicer/pull/2424

It's available in the nightly build and will be part of the upcoming 1.8 beta.

https://github.com/SoftFever/OrcaSlicer/assets/103989404/7dc05548-9f1f-4b20-9ea2-ab53a748c9fa

closed 2023-10-29 by SoftFever