OrcaSlicer/OrcaSlicer #2500: Orca Slicer not launching on Arch Linux Due to SSL Cert error.

KiwiHuman, 2023-10-23:
### OrcaSlicer Version

1.7.0

### OS version

Arch Linux

### Additional system information

OS: Arch Linux x86_64 
Kernel: 6.5.8-arch1-1 
CPU: AMD Ryzen 5 2600 (12) @ 3.400GHz 
GPU: NVIDIA GeForce RTX 3060 
Memory: 3874MiB / 15938MiB 
                           
                                                                 

### Printer

Application Not launched

### How to reproduce

1. Download .appimmage
2. launch appimmage with appimmage launcher
3. intergrate and run 
4. [image]
5. click yes or no with "remember my choice" either selected or deselected 

### Actual results

the dialog box asking about SSL certificates closes and the application dose not launch 

### Expected results

the application opens 

### Project file & Debug log uploads

There is no applicable project file or log. 
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2023-10-30:
It's system notification.
Please tick `Remember my choice` and click yes.
It will launch

closed 2023-10-30 by SoftFever

KiwiHuman, 2023-10-30:
> It's system notification. Please tick `Remember my choice` and click yes. It will launch

I selected "Remember My Choice" and yes. The dialog box closed. Relaunching OrcaSlicer just shows the dialog box again.

zetabemol, 2023-11-01:
En mi caso pasaba lo mismo. También lo tengo instalado en Windows 10. Lo que hice fue copiar la carpeta de config de orca slicer (Windows10/Users/User/AppData/Roaming/orcaslicer) de windows 10, y reemplazar la misma carpeta en archlinux (/home/user/.config/orcaslicer). De esa manera funcionó

KiwiHuman, 2023-11-02:
> En mi caso pasaba lo mismo. También lo tengo instalado en Windows 10. Lo que hice fue copiar la carpeta de config de orca slicer (Windows10/Users/User/AppData/Roaming/orcaslicer) de windows 10, y reemplazar la misma carpeta en archlinux (/home/user/.config/orcaslicer). De esa manera funcionó

Thank you so much. I had to sort out some issues with locales but this fixed it. (I used a config from Linux mint instead of windows)