OrcaSlicer/OrcaSlicer #2501: If using Filament Change, estimated Time is corrupt

tommy536, 2023-10-23:
### OrcaSlicer Version

1.7.0

### OS version

MacOS Sonoma M2

### Printer

Voron 2.4

### How to reproduce

Import STL
Adding second Filament
Slice
Go to Layer of change
Switch to second Filament
Slice
Time's corrupt

### Actual results

The Print starts normaly
If it comes to the Filament Change, print aborted with a failure

### Expected results

Normal Filament Change procedure with M600

### Project file & Debug log uploads

[Truck Robot PIP.stl.zip](https://github.com/SoftFever/OrcaSlicer/files/13072297/Truck.Robot.PIP.stl.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tommy536, 2023-10-23:
[image]

tommy536, 2023-10-24:
News: On Windows 11, with same version of Orca and STL File, the calculation is correct and the print runs till the end.

thamEnJie, 2023-10-29:
Same here, I recently had this problem. To add more information, for me, after it switches filament at the specified layer, it moves to the first position it should go to, then doesn't move.

SoftFever, 2023-10-29:
Hi
Can you share the project file?

tommy536, 2023-10-30:
It's on all STL's with a Filament Change in it. 

Unfortunatelly, the Bug is also present in 1.8 Beta.

SoftFever, 2023-10-30:
That's not helping.
Please attach a project file that can reproduce project.
The more info provided, the better you can help devs diagnose the cause.

rizwansarwar, 2023-11-04:
@SoftFever 

I just experienced the same problem, checked in 1.7.0, and 1.8 beta 2. Attached a project file where a simple color change is requested at certain layer height. I also checked Bambu Slicer, no such issue there. 

[Orca_Color_Change.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13257583/Orca_Color_Change.3mf.zip)

SoftFever, 2023-11-05:
@rizwansarwar Thanks. 
I still can't repro it with your project.
Does it happen every time?

[image]

rizwansarwar, 2023-11-05:
@SoftFever yes indeed. I suspect the problem is MAC specific and more importantly MAC Sonoma specific. I will upload the gcode generated by Orca and show some really weird extrusion where the printer hangs. Please see the attached gcode file and look at line 19496, this is where it hangs. That massive extrusion can't be right and the estimated time shown is probably because of that.

[image]

[Cube_PLA_35583992d72h49m.gcode.zip](https://github.com/SoftFever/OrcaSlicer/files/13259099/Cube_PLA_35583992d72h49m.gcode.zip)

I will try to test on Linux and see if I get the same problem, but unfortunately for now I have only got access to a MAC.

- Riz

SoftFever, 2023-11-09:
Thank you all.
I've pushed a fix for the issue.

Could you help verify if the fix works by trying the latest dev build?

SoftFever, 2023-11-09:
> @SoftFever Our wow so quick response. Where do I get the latest dev build? If you mean 1.8 beta 2 then I have already tested it and it has the same issue. If you there is some other build I can test then please let me know where can I download it? I will test and report back.

Hi @rizwansarwar 
You can download the latest dev build here: https://github.com/SoftFever/OrcaSlicer/actions/runs/6813204358

rizwansarwar, 2023-11-10:
Hi @SoftFever, I tested the build linked above on MAC M1. The problem is fixed and the slice shows correct time and there is faulty gcode extrusion. Thank you for fixing it.

SoftFever, 2023-11-11:
> Hi @SoftFever, I tested the build linked above on MAC M1. The problem is fixed and the slice shows correct time and there is no faulty gcode extrusion. Thank you for fixing it.

Thank you so much for verifying the fix and for your help in narrowing down the problem scope. 
Your assistance has been incredibly valuable!

closed 2023-11-18 by SoftFever