OrcaSlicer/OrcaSlicer #2507: Upload Files while use multiple Klipper Printers not possible

Caliban2017, 2023-10-25:
### OrcaSlicer Version

1.8.0-beta1

### OS version

Windows 11 x64 22H2

### Printer

Artilley Sidewinder X1 / Ender 3 Pro / CR10S-Pro

### How to reproduce

1 - Add a printer and add the Klipper URL in the Settings (http://klipper.local).
2 - Add another printer, which must use the same Link
3 - add generated API-Keys to the Printers

### Actual results

When using the first Printer, everything is fine. Files can be uploaded, i see the UI and can control the Printer. When i switch to the second Printer, i must manually switch the Klipper Instance to the second printer. Even with added API Key always the first printer will be opened, which might be caused by the same Link to both printers. I can not link the two printers directly with a Port at the end (http://klipper.local:7125), because this will bring a Error from Moonraker, which says: "You may have intended to navigate to one of Moonraker's front ends, if so check that you entered the correct port in the address bar."

Even if i manually switch to the second printer (http://klipper.local:7126), uploading the file is still sending the File to the first printer (http://klipper.local:7125).

### Expected results

When i chose the first Printer, the Device should switch to the first added Printer in Klipper (http://klipper.local:7125). When i switch a Profile to the second printer, it should switch to the second Klipper instance (http://klipper.local:7126).

I don't know how this should be handled, but it is not possible via the API Key for me.

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

garry54f, 2024-01-06:
I have the same problem with two printer built exactly the same but for build size

tholterhus, 2024-01-14:
Neptune 3 Pro and Anycubic Kossel on 1 Klipper instance (ports 7125 and 7127) - PITA to switch between printers :-(

ico621, 2024-03-12:
The same issue affects me. (2 printers, same IP, different ports) I can't use OrcaSlicer as my default slicer because of this detail :(

closed 2024-06-18 by github-actions[bot]