OrcaSlicer/OrcaSlicer #2509: Auxiliary fan is not starting for Flashforge printers

mariuske, 2023-10-25:
### OrcaSlicer Version

1.7.0

### OS version

Win 10

### Printer

Flashforge Adventurer 5M Pro

### How to reproduce

Using the new profiles added in #2505 

[image]

### Actual results

Using the new profiles added in #2505 the Auxiliary fan (on the left side of the chamber) does not start after the 1 layer as expected based on the configuration. It stays off for the whole print cycle.

I'm not sure if this worked with the old profiles (included on the USB drive), but I guess not based on the [following post](https://www.facebook.com/groups/913241999042183/permalink/2065201360512902/) in your offcial Facebook group:
[image]

### Expected results

Using the new profiles added in #2505 I expect the Auxiliary fan (on the left side of the chamber) to start after the 1 layer.

### Project file & Debug log uploads

[Untitled.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13168562/Untitled.3mf.zip)
[ZeroG_astronaut_PLA_29m32s.gcode.zip](https://github.com/SoftFever/OrcaSlicer/files/13168572/ZeroG_astronaut_PLA_29m32s.gcode.zip)
[debug_Wed_Oct_25_17_07_59_15664.log.zip](https://github.com/SoftFever/OrcaSlicer/files/13168591/debug_Wed_Oct_25_17_07_59_15664.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

mariuske, 2023-10-25:
@FlashforgeOfficial - Please look into this issue as this should be fixed before the new beta release (scheduled end of this week).

FlashforgeOfficial, 2023-10-26:
> @FlashforgeOfficial - Please look into this issue as this should be fixed before the new beta release (scheduled end of this week).

@mariuske Cool.  we are going to take a look and find a solution for this.

FlashforgeOfficial, 2023-10-26:
@mariuske Thank you @mariuske ! Your attention and effort are appreciated! Let's keep up the discussion for better slicing experience. 
For the fan problem, I notice this issue and contact the engineers. Below is the cause for the failure and solution:
1). Problem cause: the auxiliary fan number setting in Orca doesn't match the very same fan number as in our firmware. For instance, in Orca its M106 P2 S255, while in the firmware it's M106 P101 S255. 

2). Solution/ advice:  we will push firmware update that uses fan number P2 so that it works properly. This submitted profile will not be affected. 

By the way, I see two more discussions on the profiles, together with the tips earlier, they are good aspects of tuning, also they do need more adequate tests and conversations together. As I am currently tied up with other things for a while LOL,  let's keep track on them after the beta publish.

mariuske, 2023-10-26:
Great news @FlashforgeOfficial! Thank you for the update, changing the fan index from P101 to P2 sounds like a good solution. Do you have an estimate on the firmware update?

I think you should do a quick review of the 4 proposed changes to the 0,2mm AD5M Pro 0,4 Nozzle profile in #2514 as they should be sorted out before the release. For the the other 3 mismatches highlighted in #2514 they can be reviewed/tested later.

FlashforgeOfficial, 2023-10-27:
> Great news @FlashforgeOfficial! Thank you for the update, changing the fan index from P101 to P2 sounds like a good solution. Do you have an estimate on the firmware update?
> 
> I think you should do a quick review of the 4 proposed changes to the 0,2mm AD5M Pro 0,4 Nozzle profile in #2514 as they should be sorted out before the release. For the the other 3 mismatches highlighted in #2514 they can be reviewed/tested later.
> 
> Edit: I can confirm that if I do a find & replace of "M106 P2" with "M106 P101" in the gcode, the auxiliary fan works as expected👍

The update will be recent but I not sure about specific date neither.

Cool I am working on the changes for 0.2mm ADM5 Pro profile.  If changes are approved, I will commit changes in profile and stay updated with you. 

Nice done! The gcode helps to find where issues arise sometimes. ☺

mariuske, 2023-11-07:
This has now been resolved in firmware 2.3.6-2.1.9-20231101 by @FlashforgeOfficial and has been verified with Orca Slicer 1.8.0 Beta2.