OrcaSlicer/OrcaSlicer #2515: Support Printer Connection over Wi-Fi/Ethernet for Flashforge printers

mariuske, 2023-10-26:
**Is your feature request related to a problem? Please describe.**
Today you have to use a USB drive to transfer files to the Flashforge printers which is very cumbersome, and a big drawback compared with FlashPrint.

**Which printers will be beneficial to this feature**
Flashforge Adventurer 5M and Flashforge Adventurer 5M Pro. Possibly other models as well.

**Describe the solution you'd like**
Add support for using the Printer Connection over Wi-Fi/Ethernet functionality in Orca Slicer for Flashforge printers. Both the "send the file directly to your 3D printer" and the "Remote Printer Control" functionality.

**Describe alternatives you've considered**
The only alternatives are either using a USB drive or the Flashprint software.

**Additional context**
User documentation describing the functionality:
https://www.obico.io/blog/orcaslicer/#step-41-send-the-file-to-the-printer
https://www.obico.io/blog/orcaslicer/#remote-printer-control

It would be great if @SoftFever can share the different options available for doing such an integration with the @FlashforgeOfficial team 👍

SoftFever, 2023-10-29:
> **Is your feature request related to a problem? Please describe.** Today you have to use a USB drive to transfer files to the Flashforge printers which is very cumbersome, and a big drawback [...]

Hi
Once you’ve configured the connection(click the button with WIFI icon to setup), you can send to printer and access device interface.

mariuske, 2023-11-09:
Hi @FlashforgeOfficial. Do you have an update on the feature request? This has also been requested multiple times on your Official Facebook page.

@SoftFever - This feature request can be assigned to @FlashforgeOfficial. Thank you.

FlashforgeOfficial, 2023-11-20:
@mariuske The WIFI connection feature on Orca is not supported by Flashforge printers firmware currently. For WIFI connection, it works in FlashPrint.  I will let you know if there is any update for WIFI support.

SirVingeTheSecond, 2023-11-20:
@FlashforgeOfficial Please add support WiFi support for OrcaSlicer - Would be very appreciated!

SirVingeTheSecond, 2023-11-20:
@FlashforgeOfficial Please add WiFi support through OrcaSlicer - Would be very appreciated!

FlashforgeOfficial, 2023-12-22:
Hi all. The WIFI connection for AD5M series with OrcaSlicer will be supported in a month or so. Thanks for your support!

TheBreadGuy, 2023-12-27:
One question, will that include ethernet?

BadPope, 2024-01-19:
@FlashforgeOfficial Do you have the exact date that the update is coming? Can't wait :)

RCSchim, 2024-01-22:
> Hi all. The WIFI connection for AD5M series with OrcaSlicer will be supported in a month or so. Thanks for your support!

Today I got an auto update on my fc 5mpro, should this already include the orca WiFi printing?

eyelock-redwood, 2024-01-23:
still no Wifi printing with Orcaslicer

valeriano-manassero, 2024-01-24:
I can confirm it's still not possible to use 5M Pro ( @FlashforgeOfficial  )

RCSchim, 2024-01-25:
> Got my 5m today. Still not possible to send files with orca slicer over wifi/ethernet. :( (any updates on this?)
> 
> p.s. @FlashforgeOfficial why not add mainsail/fluid ui to klipper so we can upload files that way? Creality allready did for the k1 series an so does qidi. A lot of people will see that as a dealbreaker I'm afraid.

Totally agreed! And they already promised us exactly this - but sadly with the last firmware (few days ago) didn't see this promise being held...
Curious if Creality will implement FluidD/Mainsail from the start with their new K1C - or if we again have to wait a few weeks or month until they make up their minds ;) 
As customers we should demand these kinds of industry standards to be met!

FlashforgeOfficial, 2024-01-26:
Hi all! 
Sorry for the late, our engineers are still working this feature.  
So, it's going to take a bit longer. 

Thanks for all your support!

RCSchim, 2024-01-26:
> Hi all! Sorry for the late, our engineers are still working this feature. So, it's going to take a bit longer.
> 
> Thanks for all your support!

Great to hear from you guys! Thanks in advance - this will be an important feature!

PLEASE: it would be awesome if the printer would also TURN on fully if the power cable has power  (I mean if I use wifi socket to turn on power remotly). 
Currently your printer only goes into standby and we need to physically turn on the printer via the button.
THANKS!!!!
I promise to make a seperate video on youtube about these updates to tell more people how great your printers are (my channel: rcschim - did an inital review about the printer some weeks ago).

eyelock-redwood, 2024-01-26:
> Got my 5m today. Still not possible to send files with orca slicer over wifi/ethernet. :( (any updates on this?)
> 
> p.s. @FlashforgeOfficial why not add mainsail/fluid ui to klipper so we can upload files that way? (or respect the gpl license of klipper and release the sources) Creality allready did for the k1 series an so does qidi. A lot of people will see that as a dealbreaker I'm afraid.

i totaly agree! best thing to do for @FlashforgeOfficial  is to open up the klipper install on the printer so we can send files true mainsail/fluid.

jlodew0, 2024-01-26:
> Great to hear from you guys! Thanks in advance - this will be an important feature!
>
> PLEASE: it would be awesome if the printer would also TURN on fully if the power cable has power (I mean if I use [...]

Should be possible. The non pro lack's the button. The pcb of the motherboard on the non-pro also lacks components around the connection for the button on the motherboard. (block 6)
https://github.com/g992/flashforge-ad5m-5mpro-research/blob/main/photo/_5m_full_view_marked.png?raw=true

jlodew0, 2024-01-27:
@FlashforgeOfficial ❤️ ❤️ ❤️  --> https://github.com/FlashforgeOfficial/AD5M_Series_Klipper
I love to see we are heading in the right direction!! At first sight, it seems a bit crude still but It's still the initial upload.
Haven't tested that we can make a working build of the repo. But I allready admire that flashforge listens to it's custommers/comunity. I'm quite sure that if you release functional sources where people can work with, the customer base will grow rapidly. Certainly when social media picks this up and spread the word!

eyelock-redwood, 2024-01-29:
> @FlashforgeOfficial ❤️ ❤️ ❤️ --> https://github.com/FlashforgeOfficial/AD5M_Series_Klipper I love to see we are heading in the right direction!! At first sight, it seems a bit crude still but It's [...]

its a nice step! but still there is no way to get in the printer other then to use the touchscreen

RCSchim, 2024-02-03:
Hey all, looks like we have the opensource FW available now!!!
INFO: https://www.reddit.com/r/FlashForge/comments/1ad13kd/flashforge_released_klipper_firmware_for/
Download: https://www.flashforge.com/download-center/144
Instructions to install klipper: https://www.klipper3d.org/Installation.html
let's hope someone will make a script ;)
Is the latest firmware rooted?

eyelock-redwood, 2024-02-06:
you can send files true wifi now.! 
stil in progress i think but it works fine for me!
Info:
https://github.com/iherbak/OrcaSlicer/releases?fbclid=IwAR1KyE1_eQlgBOxUEfNy9J7aCZ3DTqRWx4vlb2DPfM6ms4W4aEJfTKZEagA

jlodew0, 2024-02-06:
> Hey all, looks like we have the opensource FW available now!!! INFO: https://www.reddit.com/r/FlashForge/comments/1ad13kd/flashforge_released_klipper_firmware_for/ Download: [...]

There's nothing more available then there was 2 weeks ago. It's just the github repo packaged in a zip file.
No config file for the ad5M/pro included, no makefile included. (see issue in the github repo --> https://github.com/FlashforgeOfficial/AD5M_Series_Klipper/issues/1) .
For now, this just seems like an effort to not violate the gpl license, nothing functional (yet I hope).

RCSchim, 2024-02-06:
> you can send files true wifi now.! stil in progress i think but it works fine for me! Info: https://github.com/iherbak/OrcaSlicer/releases?fbclid=IwAR1KyE1_eQlgBOxUEfNy9J7aCZ3DTqRWx4vlb2DPfM6ms4W4aEJfTKZEagA

Thanks a lot for the info! Didn't see this (might be hidden far down in some release notes...)
Was waiting for an Open Klipper version from Flash and then the Printertype Mainsail / Klipper would work as on my other printers. But this way is also fine (and less work).

Made a short vid about it since I was so happy how it worked! https://youtu.be/Hxi3yoAtZEY
Maybe you've not seen the webcam-> to DEVICE tab trick?  
use this in ORCA Device UI:   192.168.1.YOURIP:8080/stream_simple.html

RCSchim, 2024-02-06:
> > Hey all, looks like we have the opensource FW available now!!! INFO: https://www.reddit.com/r/FlashForge/comments/1ad13kd/flashforge_released_klipper_firmware_for/ Download: https://www.flashforge.com/download-center/144 Instructions to install klipper: https://www.klipper3d.org/Installation.html let's hope someone will make a script ;) Is the latest firmware rooted?
> 
> There's nothing more available then there was 2 weeks ago. It's just the github repo packaged in a zip file. No config file for the ad5M/pro included, no makefile included. (see issue in the github repo --> [FlashforgeOfficial/AD5M_Series_Klipper#1](https://github.com/FlashforgeOfficial/AD5M_Series_Klipper/issues/1)) . For now, this just seems like an effort to not violate the gpl license, nothing functional (yet I hope).

That would be a shady practice of them...
I thought that it's just me who's to stupid to install those Klipper files on my printer.
But with the dev version of Orca 2.0.0. that's linked below on the other commenter here, it's possible to print over wifi already...

francisconroy, 2024-02-11:
> you can send files true wifi now.! stil in progress i think but it works fine for me! Info: https://github.com/iherbak/OrcaSlicer/releases?fbclid=IwAR1KyE1_eQlgBOxUEfNy9J7aCZ3DTqRWx4vlb2DPfM6ms4W4aEJfTKZEagA

I tried to run the MacOS arm app and the OS said it was damaged and couldn't be opened. Might try to build from source and see how I go? Will there be a PR to get this into upstream?

TheBreadGuy, 2024-02-13:
> Hi all! Sorry for the late, our engineers are still working this feature. So, it's going to take a bit longer.
> 
> Thanks for all your support!

Any update on this?  maybe pre release version we can test?

gpena208777, 2024-02-13:
> I tried to run the MacOS arm app and the OS said it was damaged and couldn't be opened. Might try to build from source and see how I go? Will there be a PR to get this into upstream?

Same issues for me. No Mac support yet

inkeliz, 2024-02-16:
> Same issues for me. No Mac support yet

I'm using macOS and I compiled it, and works. Ignore the released binaries, clone the repo and compile. It will take some minutes, but works. 🎉

scottdky, 2024-02-16:
The iherbak fork fails to compile on Ubuntu Linux. It hangs the entire computer around 30% of the compile. Still anxiously awaiting this feature. Constantly moving a USB drive is getting really old.

mpolr, 2024-02-16:
@scottdky, do you try launch it under Wine?

scottdky, 2024-02-19:
Good suggestion. No joy though. I tried under Wine - failure. I tried as a Virtual Machine - OpenGL is not v2.0. Has anyone else succeded in compiling under Linux?

jlrichar, 2024-02-22:
Any update on the timing of this @FlashforgeOfficial ?  It seems that this support was removed from the nightly build.  It's OK to say it is late, I would just like to know that it is still coming.  

 Also, if you are on Discord, can you monitor the Orca server--especially the flashforge folder? 

The AD5M is an amazing printer--with a little more support this could be the new benchmark intro printer.  RN needs the full Orca support--network send to printer, and the profiles need to be flushed out more.  There is only one profile (0.2mm) for the 0.4mm nozzle, and one for the 0.6mm nozzle in Orca.  Considering the ease of switching nozzles, it would be great if there were mature profiles for all nozzles and resolutions, not to mention filament profiles for at least flashforge filaments.

FlashforgeOfficial, 2024-02-23:
> Any update on the timing of this @FlashforgeOfficial ? It seems that this support was removed from the nightly build. It's OK to say it is late, I would just like to know that it is still coming. [...]

@jlrichar Thanks for your feedback! The engineers are preparing the wifi support for Orca, might take a bit longer. For slicer profiles options, we are doing more tests, they should be set up later.

AJolly, 2024-02-24:
@FlashforgeOfficial I'd ask you to put out a complete version of your klipper source first (as needed for gpl compliance!). https://github.com/FlashforgeOfficial/AD5M_Series_Klipper/issues/1

Nickeytheblade, 2024-02-24:
@FlashforgeOfficial Yes we need to get all the source, this printer is advertised as a fully opensource klipper printer on Amazon!!

scruffyOrc, 2024-03-01:
While it would be awesome to have full Klipper source, wifi in Orca Slicer, or profiles; I'd settle for a firmware that doesn't hang at the end of prints, and a version of FlashPrint that doesn't get into a state where it no longer will send prints to the printer...if you leave FlashPrint open long enough it will eventually get into a state where you stream prints and they go, but then get really slow and never actually get to the printer. 

Their answer for both is to use Orca Slicer (with bad profiles, have you noticed that the provided profiles aren't tuned for the 5M Pro...speed, retraction and other things are not the same.) and sneaker net files to the printer.  That's not really a solution, it's a workaround (to their credit they say is just temporary until they fix the bug, but still...), was hoping to use these in the farm but for now it's just being used for prototyping things.

jnunez166, 2024-03-03:
Added the 5M Pro to my cart cause the $100 off coupon. Glad I searched OrcaSlider before purchase. Sneaker-net is not an option I have a really old 3D Printer where sneaker-net is the only option.

francisconroy, 2024-03-03:
@jnunez166 I'd never heard the term sneakernet until today, but I love it! ❤️ And I'm absolutely loving the Orca release from @iherbak, it's working well for me on my 5m. Upload is quick and reliable! Haven't been running remote print though, since I'm not sure if it does the levelling or not.

jtenniswood, 2024-03-03:
Just got the preview build working on the Mac, looks great. When will this be in the main build?

iron-bro, 2024-03-06:
@FlashforgeOfficial can you just give us an actual date for this update!! It's been dragging out a while now and the vague answers are getting really annoying now.

SonofMort, 2024-03-08:
Got my 5M Pro yesterday, waiting on the WiFi print option through Orca and I'll be over the moon with the printer so far.

SonofMort, 2024-03-08:
> > Got my 5M Pro yesterday, waiting on the WiFi print option through Orca and I'll be over the moon with the printer so far.
> 
> This one worked well for me... https://github.com/iherbak/OrcaSlicer
> 
> I'm on a Mac and had to use the command line to get it working.

I'll give it a try tomorrow, thanks for the suggestion!

3DThingy, 2024-03-09:
> > Got my 5M Pro yesterday, waiting on the WiFi print option through Orca and I'll be over the moon with the printer so far.
> 
> This one worked well for me... https://github.com/iherbak/OrcaSlicer
> 
> I'm on a Mac and had to use the command line to get it working.

Tried it on Mac but not working, although streaming is. What command line did you use?

jtenniswood, 2024-03-09:
I used the pre release version...
https://github.com/iherbak/OrcaSlicer/releases/download/V1/OrcaSlicer_A0vy.dmg

Copied that to the applications folder, then ran the command line in the
docs...

attr -dr com.apple.quarantine /Applications/OrcaSlicer.app

3DThingy, 2024-03-09:
on Intel based Mac so the ihetbak release is not supported. Nightly build x86-64  streams the camera but gives me this error 
[image]

SonofMort, 2024-03-09:
iherbak's build worked on Windows for me :)

iherbak, 2024-03-09:
Hi everybody. I just ignited the github actions. However i don't have Apple dev account i was able to build the Mac os releases without the notary test. Could somebody please test them as i have no mac's at home, and give feedback

Thanks

3DThingy, 2024-03-09:
[image]
OrcaSlicer_Mac_x86_64_V2.0.0-dev-2.dmg

iherbak, 2024-03-09:
Yes as i said i have no Apple dev cert. I don’t know how to enable running such app under MacOS. I guess there is way to trust such app.

3DThingy, 2024-03-09:
> Yes as i said i have no Apple dev cert. I don’t know how to enable running such app under MacOS. I guess there is way to trust such app.

Sorry, You right.  Once allowed in seeing it ran fine.

3DThingy, 2024-03-09:
[image]
but when uploading a file to printer

iherbak, 2024-03-09:
If the test says ok then the ip should be correct. Which machine do you have?

3DThingy, 2024-03-10:
Ip is correct and can stream the camera, but unable to upload files to the 5M Pro

iherbak, 2024-03-10:
Which firmware version do you have?

3DThingy, 2024-03-10:
2.4.5

3DThingy, 2024-03-10:
[image]
tries to transfer but stays at 0% then gets timed out

3DThingy, 2024-03-10:
[image]
did a new slice from scratch for flow rate and seems to upload ok.
Appreciate your efforts, Thank you!

YahWhoMtnDew, 2024-03-11:
> Good suggestion. No joy though. I tried under Wine - failure. I tried as a Virtual Machine - OpenGL is not v2.0. Has anyone else succeded in compiling under Linux?

Seems you have been waiting for this (and not just waiting - trying things!) for longer than I have.  If you cannot get a build under a "real" distro of linux, I don't suppose I stand a snowball's chance on my wimpy chromebook!  Sure wish it worked here - not really wanting to resort to setting up a Windoze machine just for this!  Orca talks to my Klipperfied Ender 3 just fine, so I would surely like it to work with the AD5M!  Glad I'm not the Lone Stranger thinking thusly!

stevenwalton, 2024-03-15:
> Hi all. The WIFI connection for AD5M series with OrcaSlicer will be supported in a month or so. Thanks for your support!
> Dec 21, 2023

> Hi all! Sorry for the late, our engineers are still working this feature. So, it's going to take a bit longer.
> 
> Thanks for all your support!
> Jan 25, 2024

> > Any update on the timing of this @FlashforgeOfficial ? It seems that this support was removed from the nightly build. It's OK to say it is late, I would just like to know that it is still coming.
> > Also, if you are on Discord, can you monitor the Orca server--especially the flashforge folder?
> > The AD5M is an amazing printer--with a little more support this could be the new benchmark intro printer. RN needs the full Orca support--network send to printer, and the profiles need to be flushed out more. There is only one profile (0.2mm) for the 0.4mm nozzle, and one for the 0.6mm nozzle in Orca. Considering the ease of switching nozzles, it would be great if there were mature profiles for all nozzles and resolutions, not to mention filament profiles for at least flashforge filaments.
> 
> @jlrichar Thanks for your feedback! The engineers are preparing the wifi support for Orca, might take a bit longer. For slicer profiles options, we are doing more tests, they should be set up later.
> 3 weeks ago

@FlashforgeOfficial if you Open Source (or even make Source Available) your firmware ***WE*** will fix these (and other) issues ***FOR*** you. If you're engineers are overburdened, ***WE*** are motivated enough to make the fixes. Please work with us, because we'd like to work with you.

MVerri31, 2024-03-15:
Hi, @FlashforgeOfficial I personally think you are the only ones who can compete with the dominance of Bambulab. The adventure 5M series are high-performance and well-designed machines with incredible potential (in some details even superior to the renowned BambuLab) but this is not enough if the printer is not supported on time. Unfortunately it is a tough market and wifi support for OrcaSlicer is now almost mandatory if you want to target the majority of consumers. Personally I think that open source is your trump card and I think you also think like me since you released klipper. Please do not leave the support for this print (maybe even with an "AMS" :D:D ) in order to have the only real alternative to Bambulab <3

stevenwalton, 2024-03-16:
I agree with the above. I cannot see what FF has to lose by making Source Available. I'd prefer a OS, but SA still lets us make pull requests to you. You can also do dual licensing if there is a concern about competition using your software. But let's be honest, that can be reversed (it has been in the past!) and it's not like FF isn't a "clone" itself... Unless you have proprietary software from another company you're hiding I don't see what you have to loose. 

It's 3D printers... we're makers. We want to hack on top of your projects. This is what gave Creality such an edge early on: being cheap and hackable. You have this capacity. We want you to make more money. We can do more together.

[43 more replies]