OrcaSlicer/OrcaSlicer #2527: THUMBNAIL_BLOCK_START and THUMBNAIL_BLOCK_END missed

xpert13, 2023-10-28:
### OrcaSlicer Version

1.8.0

### OS version

macOS Sonoma 14

### Additional system information

CPU: Apple M1 Pro
RAM: 16 Gb

### Printer

Creality K1

### How to reproduce

1. Create gcode file
2. Open this file via any text editor
3. Thumbnails don't have THUMBNAIL_BLOCK_START and THUMBNAIL_BLOCK_END comments

### Actual results

```
; thumbnail begin 96x96 1304
; ...
; ...
; ...
; thumbnail end
```

### Expected results

```
; THUMBNAIL_BLOCK_START
; thumbnail begin 96x96 1304
; ...
; ...
; ...
; thumbnail end
; THUMBNAIL_BLOCK_END
```

### Project file & Debug log uploads

[Cable repair bottom [apple lightning] V2.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13195996/Cable.repair.bottom.apple.lightning.V2.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SoftFever, 2023-10-30:
@xpert13 
Thank you for reporting it.
It has been fixed.
This build will contain the fix: https://github.com/SoftFever/OrcaSlicer/actions/runs/6694668366

discip, 2023-11-17:
Sorry for being late, but unfortunately this breaks the thumbnail creation for BTT TFTs.

The extra line at the beginning and between thumbnails prevents this from working.

Maybe someone could change this so that it only applies to QOI. 😅

Rsoft1976, 2023-12-08:
But these lines don’t work for me, if you take the plugin from CURA, then these lines are not there

xpert13, 2023-12-08:
@Rsoft1976 It's depend on printer I guess. My Creality K1 doesn't want to show preview without those lines.

Rsoft1976, 2023-12-08:
the screen costs BIGTREETECH TFT35 V3.0 and for some reason it starts working if there are 3 resolutions 70x70,95x80, 95x95, if you remove one resolution it stops showing on the screen

Rsoft1976, 2023-12-08:
need to make a separate choice to install or not these lines in the form of a choice just below