OrcaSlicer/OrcaSlicer #2528: [Feature Request] Restart Failed Print using existing 'Cut' feature

Ltek, 2023-10-28:
**Is your feature request related to a problem? Please describe.**
We need an easy way to restart a failed print when the print is still on the build plate in proper position
-- this happens often when the fail is not catastrophic and the print is fine / intact on the build plate.
We need to restart at the last good layer... which we can find by measuring what is on the plate.

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
Use the existing 'Cut' feature, tell Orca to "Restart Print at Upper Part Cut Height" and it would SKIP XYZ Homing and using the existing parameters to restart the print at the height specified in the Cut for 'Upper Part'.

**Describe alternatives you've considered**
The only other option is to manually edit g-code which is tedious and unreliable (prone to human error)... and still have a a Homing problem if the print is still on the plate.

**Additional context**
Screenshot...
[image]

Ocraftyone, 2023-11-01:
I do not see this happening as most printers/printer firmwares require homing before printing. Also, most modern firmwares have a way to do this already

Ocraftyone, 2023-11-01:
Happy to help!  Here is a macro for klipper (https://github.com/The--Captain/plr-klipper) and Marlin docs (https://marlinfw.org/docs/gcode/M413.html). The feature is called power loss recovery.

closed 2024-02-21 by github-actions[bot]

zombodotcom, 2025-05-02:
can we bring this issue back up