OrcaSlicer/OrcaSlicer #2530: Minimum layer time limit throttling produce inconsistent speeds on identical layers seemingly randomly

tastyratz, 2023-10-28:
### OrcaSlicer Version

1.7

### OS version

Windows 10

### Printer

Voron Trident

### How to reproduce

Put a model on the plate, even a very simple model. In my example, I used 2020 extrusion and resized the model up high so it should be the same from the first to last layer. Increase acceleration and print speeds higher than the minimum layer time can support.

### Actual results

Layer time stays reading as consistent, but, print speeds are wildly different. There is visible banding in the preview and print. 
It appears in an attempt to throttle to meet layer time minimums each layer has different results.

There is no discernable pattern on the preview to the speeds all the way up. It is wholly inconsistent
[image]

### Expected results

The algorithm for minimum layer time does not appear to produce consistent results OR compare the changes to the following layer. 
For more consistent prints there should be some lookahead. Perhaps layer throttling changes should be smoothed in some way for more consistent prints.

Also, it may be worth investigating why identical layers are producing very different results.

### Project file & Debug log uploads

This can be replicated with a simple cylinder if it's small enough with high enough print speeds.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

closed 2024-02-13 by github-actions[bot]

tastyratz, 2024-02-13:
Well, that's unfortunate...

dewi-ny-je, 2024-06-03:
Try attaching a 3mf project showing the issue, or try to reproduce with a plain cube, or other similarly homogeneous models which should print at the same speed but they don't.

igiannakas, 2024-06-05:
I've recently raised a PR with improvements to the handling of minimum layer time here:

It maintains a consistent external perimeter print speed so you dont get what looks like z banding at your parts due to the speed variations.
https://github.com/SoftFever/OrcaSlicer/pull/5148