OrcaSlicer/OrcaSlicer #2532: Feature: Add option for direction of concentric bottom infill

knw257, 2023-10-29:
**Is your feature request related to a problem? Please describe.**
When using concentric infill on the first layer, the pattern starts at the middle of the area and moves outward. This often causes adhesion issues, as the area starts small, making it easier for the nozzle to catch and detatch already laid down material

**Describe the solution you'd like**
Currently PrusaSlicer is configured to start a concentric pattern from the walls moving inward. Having the option regarding which direction the concentric infill is drawn would be helpful, in case the center-outward pattern is useful for some, and the wall-inward is useful for others.

**Describe alternatives you've considered**
The only solution I've found to this is to use a different slicer (i.e. PrusaSlicer)

**Additional context**
Screenshots comparing the first layer concentric infill gcode preview from both OrcaSlicer and PrusaSlicer

[image]
[image]

knw257, 2023-11-17:
It appears that in the latest release, 1.8.0, the concentric direction has been changed to start from the walls and proceed inward. Though this change is not listed in the release notes that I can see, it is a welcome change. I still believe having the option might be useful for some, though I do not know in which situations this might be useful, but I imagine it was originally configured to start from the middle and proceed outward for a reason.

In any case, my thanks to whomever made the change in the latest release.

knw257, 2023-11-22:
Turns out I was wrong - at least partially. It seems while some models now have their first layer concentric infills printed from the walls to the center, others do not in OrcaSlicer. As such, I'm reopening this issue in hopes that the behavior can be made more consistent.

closed 2024-02-29 by github-actions[bot]

dstulken, 2024-03-05:
> I still believe having the option might be useful for some, though I do not know in which situations this might be useful, but I imagine it was originally configured to start from the middle and proceed outward for a reason.

The two directions would have significantly different results if the first layer was over-extruded in any way (either due to a z-offset too close or a flow miscalibration).  

With the "start from the outside" path, the outer portion would be fine, but the nozzle would progressively push a growing mound of excess filament toward the center that could crud up the nozzle or interfere with the layers printed above.   

With the "start from the inside" path, the central portion would be fine, but it would likewise start pushing excess filament - this time outward.  However, since there is more space for this filament to get pushed to, it would likely just expand the first layer slightly - leaving a smooth surface for the second layer to print upon.

Note that Cura went down the same path with concentric printing and ironing - ultimately chosing to print from inside-out for the best surface quality.
https://github.com/Ultimaker/Cura/issues/2390

If there is any way to re-open this issue, I would second the request to allow the direction to be configurable.

knw257, 2024-03-05:
> If there is any way to re-open this issue, @knw257, I would second the request to allow the direction to be configurable.

Unfortunately, I don't seem to have the ability to re-open the issue, I think SoftFever or one of the main contributors to the repo have to do that. Hoping they will, because I'd very much appreciate the option on this

pegurov, 2025-04-27:
I have found the most accurate way to calibrate flow on printables -
https://www.printables.com/model/538734-improved-flow-rate-calibration
It works on the principle of progressively pushing excess filament inwards, making even the slightest overextrusion very obvious.

However, it requires concentric pattern to be printed from outside to the inside, which makes it unreliable, because I cannot manually control concentric infill direction. Right now if I place several of the test patterns on one plate, some of them are printed from outside in, others in two steps, in a weird pattern.

Having the option to manually control concentric pattern direction would really help with faster flow calibration using above pattern.

[image]

evilDave, 2025-05-21:
Yeah. This is pretty annoying. On the one model it chooses inside to outside AND outside to inside, even on different almost identical features.
I have made a simple calibration target that is basically just 10mm circles at each of the mesh probing points. I want to print them concentric because that seems to show up over/under extrusion pretty well. However some of them are inside out and other outside in.
Some consistency would be good, as would a way to change the order.

I have not opened another issue as this issue was closed as "won't fix" so I feel the correct way forward is to reverse that decision and re-open this ticket. (I see my least favourite git bot, the stale ticket closer is responsible, not an actual person, so maybe we can get this looked at and someone may consider it worthwhile)