OrcaSlicer/OrcaSlicer #2537: PA Pattern on Delta printers - Vector too long

cpasoft, 2023-10-29:
### OrcaSlicer Version

Orca 1.8-Beta1

### OS version

Windows 10

### Printer

FLSUN V400

### How to reproduce

We create a PA calibration in Pattern mode. When we try to laminate we get an error.

If we place the cube on the edge, it is able to laminate it, but it is impossible to place it in the centre.

I have uploaded a video that explains this better.

https://github.com/SoftFever/OrcaSlicer/assets/29692062/dad42102-9e33-4fc6-bc8f-3e4a387ec327

### Actual results

SEE VIDEO

### Expected results

Ideally, the pattern should be laminated in the centre of the bed.

### Project file & Debug log uploads

I've upload a video.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2023-11-06:
@cpasoft Can you share the project file?

cpasoft, 2023-11-06:
I'll upload it for you, of course.
Anyway, just select a V400 printer, create a new project and then select the PA pattern in the calibration menu.

Give me an hour and I'll upload it...

SoftFever, 2023-11-06:
> I'll upload it for you, of course. Anyway, just select a V400 printer, create a new project and then select the PA pattern in the calibration menu.
> 
> Give me an hour and I'll upload it...

When you got time, can you try this build?  https://github.com/SoftFever/OrcaSlicer/actions/runs/6769494046
I've fixed an vector too long bug, very likely it's the same bug.

cpasoft, 2023-11-06:
> When you got time, can you try this build? https://github.com/SoftFever/OrcaSlicer/actions/runs/6769494046 I've fixed an vector too long bug, very likely it's the same bug.

I'll try it right now...

for the moment I upload the project you asked me for
[Pressure Advance Test.zip](https://github.com/SoftFever/OrcaSlicer/files/13265508/Pressure.Advance.Test.zip)

cpasoft, 2023-11-06:
> When you got time, can you try this build? https://github.com/SoftFever/OrcaSlicer/actions/runs/6769494046 I've fixed an vector too long bug, very likely it's the same bug.

Now it no longer gives an error, although it still does strange things. I'll upload a video.
If you move the cube to the edges, the pattern also moves to the opposite side. If you centre it, it looks more or less OK, but if you put it on the right side of the bed, the cube goes over the test.
It's functional now, but it still does strange things.

https://github.com/SoftFever/OrcaSlicer/assets/29692062/97ca53e7-e9fa-4ae4-b85a-95f8dcbd2868

SoftFever, 2023-11-06:
> > > I'll upload it for you, of course. Anyway, just select a V400 printer, create a new project and then select the PA pattern in the calibration menu.
> > > Give me an hour and I'll upload it...
> > 
> > 
> > When you got time, can you try this build? https://github.com/SoftFever/OrcaSlicer/actions/runs/6769494046 I've fixed an vector too long bug, very likely it's the same bug.
> 
> Now it no longer gives an error, although it still does strange things. I'll upload a video. If you move the cube to the edges, the pattern also moves to the opposite side. If you centre it, it looks more or less OK, but if you put it on the right side of the bed, the cube goes over the test. It's functional now, but it still does strange things.
> 
>  Rec.0009.mp4

Thanks for the feedback.
Fixed  in main branch

cpasoft, 2023-11-06:
Thank you as always, you are doing a great job, and we love it!

cpasoft, 2023-11-06:
Now it is working perfect. Thanks again...
[image]