OrcaSlicer/OrcaSlicer #2538: Aux_fan no longer works

SlackHacky, 2023-10-29:
### OrcaSlicer Version

1.8.0. beta 

### OS version

OS windows 11

### Printer

VzBot 330

### How to reproduce

n/a 

### Actual results

Aux_fan no longer works - rolling back to orca 1.7.0 is works again, with no change of settings - so very 1.8.0 related rather than i've got a setting wrong.

### Expected results

aux_fan turning on with parts cooling fan at pre set level

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [X] Log file
- [ ] Project file

SoftFever, 2023-10-30:
@SlackHacky 
Can you share the project file?

diezjl, 2024-01-13:
I have the same issue since version 1.8.0, in version 1.9.0 still persist.
I realised some test and see that the issue only occurs when select auxiliarity fan speed below 100%, in this cases the g-code generated miss "M106 P2 Sxx" command.

SlackHacky, 2024-01-13:
can confirm no longer works as was using Rigid static cooling as Aux cooling on a vzbot and since 1.8.0 only woks at 100% not 99% as worked for 1.7.0

diezjl, 2024-03-18:
The problem persist in 2.0.0 beta

closed 2024-08-09 by github-actions[bot]